apache/plc4x #1582: [Bug]: Building PLC4C on AARCH64 Linux machines seems to be broken

chrisdutz, 2024-05-03:
### What happened?

When trying to build PLC4C on a VM running Ubuntu Linux on my Aarch64 computer (Inside Parallels) the build fails for some reason that seems to come from CMake itself.

### Version

v0.13.0-SNAPSHOT

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [X] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-05-05:
Built a workaround, that uses the dependency plugin go fetch an artifact with the classifier windows-arm-64 and locally install it as windows-arm64 ... seems to make the build work.

sruehl, 2024-05-06:
hmm, this ticket is about linux and you write something about windows?

sruehl, 2024-05-06:
linux builds for me on 5cbd00583b85bfee43d7e8128dcda4ab0c14c228 so IMHO this ticket is good to be closed

chrisdutz, 2024-05-06:
Well it doesn#t build for me ... so we need to find out what's going on

chrisdutz, 2024-05-06:
I built the same workaround (well allmost) to the one in windows ... so I'm downloading the aarch64 binaries from the cmake site and repackaging them as jar.

closed 2024-05-24 by chrisdutz