apache/plc4x #1583: [Bug]: Building PLC4Py on AARCH64 Windows machines seems to be broken

chrisdutz, 2024-05-05:
### What happened?

Problems building

### Version

v0.13.0-SNAPSHOT

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

hutcheb, 2024-05-05:
Might need a bit more information.

chrisdutz, 2024-05-05:
Just wrote that down as a marker ... it seems that Python seems to need VisualStudio C++ Compiler, but tries to run the x86_64 version instead of the arm64 version ... was planning on looking into this friday in 2 weeks, when I've got my next 4h of PLC4X.

closed 2024-05-24 by chrisdutz