apache/plc4x #1589: [Bug]: org.apache.plc4x.java.api.exceptions.PlcProtocolException: Server returned error BadTooManySessions (0x80560000)

ShawnKion, 2024-05-07:
### What happened?

When I tried to read some nodes multiple times, the issue of too many sessions occurred.
I'm wondering if there is a configuration file that I can set some values to limit the max session I need?

### Version

0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

splatch, 2024-05-24:
Hello @ShawnKion,
Can you share some more information - how do you scrap PLC and how long it does take to get this error?

Strictly speaking it is runtime issue, so we have to look closer at how you read data and, probably, what are server settings.

qtvbwfn, 2024-08-10:
@ShawnKion 
The OPCUA SERVER limits the number of connection sessions, so make sure you haven't established too many connections. You also need to check the OPCUA SERVER's limitations regarding connections.

splatch, 2024-10-22:
This issue is most likely environment specific. Make sure that you 1) open few connections to PLC 2) construct application logic independently of connection 3) manage number of opened connections according to hardware manufacturer limits.

I am closing it for now - please re-open it if you will be able to provide more details or suspect bug in driver itself. You can try running extra diagnostics to see if your app is able to keep connection active for long time.

closed 2024-10-22 by splatch