apache/plc4x #1593: [Bug]: S7 connection problem

korni128, 2024-05-07:
### What happened?

The PlcConnection is not working propertly in my setup for S7 PLC. Maybe it is related to Issue 1257.

My test code (the test is not successful).
```
import static org.assertj.core.api.Assertions.assertThat;
import org.apache.plc4x.java.DefaultPlcDriverManager;
import org.apache.plc4x.java.api.PlcConnection;
import org.junit.jupiter.api.Test;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

public class BasicConnectionTest {

  @Test
  void testConnection() throws Exception {
    String url = "s7://127.0.0.1:102?remote-rack=0&remote-slot=1";
    PlcConnection plcConnection = new DefaultPlcDriverManager().getConnectionManager().getConnection(url);
    assertThat(plcConnection.isConnected()).isTrue();
    plcConnection.close();
  }
}
```
Apart from that, some requests seem to work properly when the connection is used afterwards.

Log output produced with test code in Version v0.12.0
```
 INFO  o.a.p.java.DefaultPlcDriverManager - Instantiating new PLC Driver Manager with class loader jdk.internal.loader.ClassLoaders$AppClassLoader@76ed5528
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering available drivers...
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol s7 (Siemens S7 (Basic))
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol plc4x (PLC4X (Proxy-Protocol))
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol mock (Mock Protocol Implementation)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol ab-eth (Allen Bradley ETH)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol ads (Beckhoff TwinCat ADS)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol bacnet-ip (BACnet/IP)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol c-bus (Clipsal C-Bus)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol genericcan (Generic CAN)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol canopen (CAN open)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol eip (EthernetIP)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol logix (Logix CIP)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol firmata (Firmata)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol iec-60870-5-104 (IEC 60870-5-104)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol knxnet-ip (KNXNet/IP)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol open-protocol (Open-Protocol)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol modbus-ascii (Modbus ASCII)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol modbus-rtu (Modbus RTU)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol modbus-tcp (Modbus TCP)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol opcua (Opcua)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol profinet (Profinet)
 INFO  o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol simulated (Simulated PLC4X Datasource)
 INFO  o.a.p.j.t.tcp.TcpChannelFactory - Configuring Bootstrap with org.apache.plc4x.java.s7.readwrite.configuration.S7TcpTransportConfiguration@78a287ed
 INFO  o.a.p.j.s.r.protocol.S7ProtocolLogic - S7 Driver running in ACTIVE mode.
 INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - 15:19:00.091903500 userEventTriggered: Multiplexor Event: org.apache.plc4x.java.spi.events.ConnectEvent@6bc248ed
 WARN  i.n.channel.embedded.EmbeddedChannel - More than one exception was raised. Will report only the first one and log others.
io.netty.handler.codec.DecoderException: io.netty.handler.codec.EncoderException: MessageToMessageCodec$1 must produce at least one message.
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:98)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:103)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:346)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:318)
	at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
[... 232 lines cut ...]
### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

glcj, 2024-05-07:
Hello, Just to discard

If it is an S7-300, it should be String url = "s7://XXX.XXX.XXX.XXX.:102?remote-rack=0&remote-slot=2";
If it is an S7-400,  String url = "s7://XXX.XXX.XXX.XXX:102?remote-rack=0&remote-slot=3";

So, Are you  using NetToPLCSim or something similar? I say it because of the IP address is a local PC.

Regards,

korni128, 2024-05-08:
Hello César José García León, 

thank you for looking into the problem and responding. 

We have three real S7 PLCs in our test lab:: S7-1200, S7-1515, S7-1516
For testing purposes, I connect to the PLC via an SSH tunnel, but I also run the code without an SSH connection next to the PLC. I have improved my code and logging a little bit to show you more details. I have tried out the connection with all three SPS via SSH and next to the SPS on a remote computer. Everything runs perfectly in version 0.11.0 but if I change the version to 0.12.0 the exception occurs as described before. You can find more details in the log output below. 

Here is my current test code:
```
import org.apache.plc4x.java.api.PlcConnection;
import org.apache.plc4x.java.api.PlcDriverManager;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

public class TestConnectionWithPLC4X2 {
  
  private static final Logger logger = LoggerFactory.getLogger(TestConnectionWithPLC4X2.class);

  public static void main(String[] args) throws Exception {

    String urlS71515 = "s7://192.168.0.111:102?remote-rack:0&remote-slot:1"; // S7-1515 (Lab)
    String urlS71200 = "s7://192.168.0.190:102?remote-rack:0&remote-slot:1"; // S7-1200 (Error_FBs)
    String urlS71516 = "s7://192.168.0.186:102?remote-rack:0&remote-slot:1"; // S7-1516 (Simulation)

    String executionEnvironment = System.getenv("EXEC_ENVIRONMENT");
    if ("local".equals(executionEnvironment)) {
      urlS71515 = "s7://127.0.0.1:1515?remote-rack:0&remote-slot:1"; // S7-1515 (Lab)
      urlS71200 = "s7://127.0.0.1:1200?remote-rack:0&remote-slot:1"; // S7-1200 (Error_FBs)
      urlS71516 = "s7://127.0.0.1:1516?remote-rack:0&remote-slot:1"; // S7-1516 (Simulation)
    }

    logger.info("");
    logger.info("#####################################################################");
    logger.info("Test connection with S7-1515");
    logger.info("#####################################################################");
    try (PlcConnection connection = PlcDriverManager.getDefault().getConnectionManager().getConnection(urlS71515)) {
      logger.info("Connection status with S7-1515: {}", connection.isConnected() ? "CONNECTED" : "DISCONNECTED");
    }
    logger.info("Connection Closed");

    logger.info("");
    logger.info("#####################################################################");
    logger.info("Test connection with S7-1200");
    logger.info("#####################################################################");
    try (PlcConnection connection = PlcDriverManager.getDefault().getConnectionManager().getConnection(urlS71200)) {
      logger.info("Connection status with S7-1200: {}", connection.isConnected() ? "CONNECTED" : "DISCONNECTED");
    }
    logger.info("Connection Closed");

    logger.info("");
    logger.info("#####################################################################");
    logger.info("Test connection with S7-1516");
    logger.info("#####################################################################");
    try (PlcConnection connection = PlcDriverManager.getDefault().getConnectionManager().getConnection(urlS71516)) {
      logger.info("Connection status with S7-1516: {}", connection.isConnected() ? "CONNECTED" : "DISCONNECTED");
    }
    logger.info("Connection Closed");
  }
}
```

Log output with version 0.11.0 => works and looks good
```

 [main] INFO TestConnectionWithPLC4X - #####################################################################
 [main] INFO TestConnectionWithPLC4X - Test connection with S7-1515
#####################################################################
 [main] INFO o.a.p.java.DefaultPlcDriverManager - Instantiating new PLC Driver Manager with class loader jdk.internal.loader.ClassLoaders$AppClassLoader@76ed5528
 [main] INFO o.a.p.java.DefaultPlcDriverManager - Registering available drivers...
 [main] INFO o.a.p.java.DefaultPlcDriverManager - Registering driver for Protocol s7 (Siemens S7 (Basic))
[... 728 lines cut ...]
	at java.base/java.util.function.Consumer.lambda$andThen$0(Consumer.java:65)
	at org.apache.plc4x.java.spi.Plc4xNettyWrapper.decode(Plc4xNettyWrapper.java:183)
	at io.netty.handler.codec.MessageToMessageCodec$2.decode(MessageToMessageCodec.java:81)
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:88)
	... 40 common frames omitted
 [nioEventLoopGroup-4-1] DEBUGDOOM - [id: 0xembedded, L:embedded - R:embedded] READ: 967B
         +-------------------------------------------------+
         |  0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f |
+--------+-------------------------------------------------+----------------+
|00000000| ..............
 [nioEventLoopGroup-4-1] DEBUGDOOM - [id: 0xembedded, L:embedded - R:embedded] USER_EVENT: org.apache.plc4x.java.spi.events.ConnectedEvent@4ec2b1c8
 [nioEventLoopGroup-4-1] INFO o.a.p.j.s.r.protocol.S7HMuxImpl - 15:07:58.400353500 userEventTriggered: Multiplexor Event: org.apache.plc4x.java.spi.events.ConnectedEvent@4ec2b1c8
 [main] INFO TestConnectionWithPLC4X - Connection status with S7-1516: DISCONNECTED
 [nioEventLoopGroup-4-1] INFO o.a.p.j.s.r.protocol.S7HMuxImpl - java.util.NoSuchElementException: watchdog
 [main] DEBUGDOOM - [id: 0xembedded, L:embedded - R:embedded] USER_EVENT: org.apache.plc4x.java.spi.events.DisconnectEvent@6dee4f1b
 [main] DEBUGDOOM - [id: 0xembedded, L:embedded - R:embedded] CLOSE
 [main] DEBUGDOOM - [id: 0xembedded, L:embedded ! R:embedded] INACTIVE
[... cut ...]
 [main] INFO o.a.p.j.s.r.protocol.S7HMuxImpl - Unregistered of channel: PRIMARY
 [main] INFO o.a.p.j.s.r.protocol.S7HMuxImpl - 15:07:58.400353500 userEventTriggered: Multiplexor Event: org.apache.plc4x.java.spi.events.DisconnectEvent@6dee4f1b
 [main] INFO o.a.p.j.s.r.protocol.S7HMuxImpl - DisconnectEvent
 [main] INFO TestConnectionWithPLC4X - Connection Closed
 [plc4x-evt-processor-thread-1] INFO o.a.p.j.s.r.p.S7ProtocolEventLogic - ObjectProcessor Bye!
 [plc4x-evt-dispatcher-thread-1] INFO o.a.p.j.s.r.p.S7ProtocolEventLogic - EventDispatcher Bye!
```

mrwhy-orig, 2024-05-16:
I'm facing the same Exceptions, but the connections working fine. I'm able to read and write data as well as getting notifications.

korni128, 2024-05-16:
Hi [mrwhy-orig](https://github.com/mrwhy-orig),

thank you for your comment. Can you also take a look at the connection status after the connection has been established. Is it connected in your case? Which PLC version are you using?

glcj, 2024-05-16:
Hello,

This exception is generated internally by Netty when initiating the connection and should not affect the general operation of the driver.

Greetings,

korni128, 2024-05-16:
Hello [glcj](https://github.com/glcj),

you are right it is possible to perform read requests with the connection. I have not tested other operations but [mrwhy-orig](https://github.com/mrwhy-orig) commented that it is possible. I also mentioned in the issue description that the connection requests seem to work.

> Apart from that, some requests seem to work properly when the connection is used afterwards.

My problem is that when I call plcConnection.isConnected() it is always false in version 0.12.0.
This already affects my current implementation as I check the connection before a new request is executed. If the PLC connection is diconnected, I try to re-establish the connection.

I'm not sure about following log but it says that more then one exception was raised.
```
[nioEventLoopGroup-4-1] WARN i.n.channel.embedded.EmbeddedChannel - More than one exception was raised. Will report only the first one and log others.
io.netty.handler.codec.DecoderException: io.netty.handler.codec.EncoderException: MessageToMessageCodec$1 must produce at least one message.
```

The second problem is the log: I have tried to change the log level of PLC4X and netty to ERROR but the DEBUGDOOM stuff is still present in the log. 

It seems that in my case there is no good workaround with version 0.12.0.

chrisdutz, 2024-05-17:
If I do a search for "S7HMuxImpl.IS_CONNECTED" I get only reading results ("channel.attr(S7HMuxImpl.IS_CONNECTED).get()" and not a single set ... so I would assume that's the reason, why "isConnected" always returns false (It checks this in order to say if the driver is connected or not)

I just ran my ManualTest against my real S7 and besides the multi-item-write bug that I was planning on working on today, that all worked perfectly. But opening a new bug for the "isConnected" stuff .. that should work.

chrisdutz, 2024-05-17:
I just double-checked and my last assumption is actually not true ... it's not always returning false.

korni128, 2024-05-17:
I had a look into version 0.10.0 and 0.11.0 As far as I can follow the code: the method isConnected from DefaultNettyPlcConnection was not overwritten by S7HPlcConnection. So maybe you just need to completely remove the isConnected method in S7HPlcConnection?

Apart from the isConnected method, I would say that the WARN and DEBUGDOOM logging from the nioEventLoopGroup process is still problematic. It seems to work, but it messes up all the logging and important messages can get lost.

glcj, 2024-06-04:
Hello,

Perform the tests with 0.13.0-SNAPSHOT against an S7-1200 (1214C) without problems connecting or reading data.

Use the same example for S7-1200 available in the repository.

Kind regards,

glcj, 2024-06-04:
Hello,

Perform the tests with 0.13.0-SNAPSHOT against an S7-1200 (1214C) without problems connecting or reading data.

Use the same example for S7-1200 available in the repository.

If you agree we can close this issue

Kind regards,

korni128, 2024-06-04:
Hello [glcj](https://github.com/glcj),

I have tested my sample code with version 0.13.0-SNAPSHOT (0.13.0-20240531.145431-1). Still the same behavior: isConnected is false after the connection has been established and the DEBUGDOOM and WARN log is still present.
Was there a related commit? 
I have tried out to remove the isConnected method in the S7HPlcConnection class and then the connection status is "true". But not sure why this stuff is required (introduced in version 0.12.0 with #1214):
```
  @Override
  public boolean isConnected() {
    return channel.attr(S7HMuxImpl.IS_CONNECTED).get();
  }
```
I also think that the DEBUGDOOM problem is a bit more difficult to debug / solve, at least for me, because there were some time constrains during my attempts.

chrisdutz, 2024-06-04:
Search for "BOOM"... It's a decorator for log statements, I think

chrisdutz, 2024-06-30:
I commented out the BOOM LoggingHandler ... so that should vanish from the logs.

splatch, 2024-06-30:
I've checked develop with s7-1200 and call to `isConnected` did not produce null pointer. I had no issues with it. Looks like it works out of the box.
One reason I could think of, why NPE shows, is different version of netty used at runtime. I still need to confirm that / double check output of isConnected.

chrisdutz, 2024-06-30:
Yeah ... have tested it with my S7-1200 models as well as my S7-1500 and I also couldn't see any issues.

chrisdutz, 2024-06-30:
I think while investigating another problem I stumbled over what's causing the "must produce at least one message" error message. And it turns out I just didn't see the error with my test-cases, as I configured the logging to only report errors and this was a warning. As soon as I reduced the log-level to "warn" I could also see the error message.

It seems in S7HMuxImpl.encode when sending the second package (The S7 Setup Communication Request), the method bypasses the Netty pipeline and directly sends out the packet via the tcp-channel ... so the surrounding channel thinks there's nothing going out and logs a message.

I don't think that writing something to the output in a method called "encode" is a good choice.

I think this part of the driver would probably need a major refactoring. One that's currently out of scope for me, as I simply don't have enough time to work on.

doleron, 2024-07-05:
I agreed with @chrisdutz . I'm experiencing the same issue of `io.netty.handler.codec.EncoderException` flooding. Meanwhile, my workaround was to silence all WARNs like follows:

```java
        public static void main(String[] args) {
		
		System.setProperty(SimpleLogger.DEFAULT_LOG_LEVEL_KEY, "Error");
	        Logger logger = LoggerFactory.getLogger(Main.class);
		
		String connectionString = "s7://192.168.26.39?remote-rack:0&remote-slot:2&controller-type:S7_300";
		PlcConnection plcConnection = null;
		
		try {
			
			plcConnection = PlcDriverManager.getDefault().getConnectionManager()
					.getConnection(connectionString);

                       //...

```

splatch, 2024-08-09:
@doleron the connection uri looks wrong in your example, it uses `:` instead of `=`. Are you still facing connection errors?

korni128, 2024-08-27:
I have tried the latest snapshot version 0.13.0-20240405.095431-1 with the code I provided initially. The function 'connection.isConnected()' is true after the connection is established, and when I simulate a disconnection, it is false. Also, the connection is automatically re-established, and isConnect() returns true after the connection is re-established. So I would say this is working fine. If I set the Netty log level to OFF, the log is not problematic either. It's still weird, but as @chrisdutz said, it's a major refactoring and as long as it doesn't affect stability, I'd say we can try version 0.13.0 when it's available.

korni128, 2024-08-27:
> would it be ok for us to close this issue then?

@chrisdutz sure. Thank you for investigating the problem and finding a solution.

closed 2024-08-27 by chrisdutz