apache/plc4x #1602: [Bug]: EIP only working with variables with an odd character count (in second path segment)

luedi128, 2024-05-16:
### What happened?

We have a Rockwell function block that is offering the following variables, that we want to read out.

The following request with eight variables showed that only one of them is working - the other caused INTERNAL_ERROR responses from the Rockwell PLC. 

      readRequestBuilder = plcConnection.readRequestBuilder();
      readRequestBuilder.addTagAddress("v1", "xxx1.timestamp_xxxxxxx_OFF");   // ok - working
      readRequestBuilder.addTagAddress("v2", "xxx1.timestamp_xxxxxxx_ON"); // INTERNAL_ERROR

We have had a look at the CIP packages on network level and it looks like the size of the second path segment was wrongly calculated - the following part of the CIP with slot/address are not recognized as a valid part of CIP protocol in wireshark (screenshot attached). 

For testing we renamed the last variable by adding an additional character at the end (in the FB and in the java code) e.g. xxx1.timestamp_xxxxxxx_ONN then everything worked fine. We have this situations with other variables as well (8 variables in our test - only have of them (the once with an even character count) are working - the others result in INTERNAL_ERRORs)

We have been able to fix the problem by changing org.apache.plc4x.java.eip.base.protocol.EipProtocolLogic.class in line 692 to:
 newSegment = new DataSegment(new AnsiExtendedSymbolSegment(identifier, (identifier.length() % 2 == 0) ? null : (short) 0));
More or less copy & paste from line 696/697.

From out point of view this is a bug in the path segment length calculation only affecting path variables with an odd character count (or second path segments with an odd character count).

Tested on a Rockwell 1769-L18ER/B LOGIX5318ER.

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

luedi128, 2024-05-16:
Wireshark screenshot.

[image]

chrisdutz, 2024-07-12:
Could you please try with the current SNAPSHOT version ... I am using the driver in my testsuite to talk to a real device and have both variables with and without odd lengths and it's working fine.

luedi128, 2024-07-18:
I will get in touch with the Rockwell developer to provide the test setup again. Will test it then. 
Still it only happens with a segmented path. What are the names of your test variables?

chrisdutz, 2024-07-26:
So ... any update? Otherwise I'd close this issue next time I work on it (Probably next Friday)

luedi128, 2024-07-26:
It will take some time until I have the test setup available and can test again (approx. 8 weeks).  

What are the names of your test variables?

chrisdutz, 2024-07-26:
```
ManualEipLogixDriverTest test = new ManualEipLogixDriverTest("logix://192.168.23.40");
        // This is the very limited number of types my controller supports.
        test.addTestCase("hurz_BOOL", new PlcBOOL(true));
        test.addTestCase("hurz_SINT", new PlcSINT(-42));
        test.addTestCase("hurz_INT", new PlcINT(-2424));
        test.addTestCase("hurz_DINT", new PlcDINT(-242442424));
        test.addTestCase("hurz_REAL", new PlcREAL(3.141593F));
        //test.addTestCase("hurz_UDT", new PlcStruct());

        long start = System.currentTimeMillis();
        test.run();
        long end = System.currentTimeMillis();
        System.out.printf("Finished in %d ms", end - start);

```

chrisdutz, 2024-07-26:
As you can see, it contains both types of lengths (odd/even)

luedi128, 2024-07-26:
Yes that worked for us as well. The situation only appears with segmented variables (variable names with a dot in the name, like a variable in a function block)
xxx1.timestamp_xxxxxxx_ON

chrisdutz, 2024-07-26:
I think I had a look at this and the main reason is, that I think the version of the driver in the open-source project currently only supports (not sure if this is the right term) Global Variables that are not located in a function block.

However it is generally possible to read them, as I have done a private experiment with a completely rewritten version of the Logix driver and here I have tested this feature. Unfortunately I will not donate this work back without reasonable donations, which I described the reasons for in my last blog post.

luedi128, 2024-08-15:
Now I have the test environment available again. 
I can still reproduce the situation with these two variables:
xxx1.timestamp_xxxxxxx_ON
xxx1.timestamp_xxxxxxx_OFF

How can I get a Snapshot-Version? You refer to 0.13.0-SNAPSHOT right? Are there compiled snapshot artifacts available?

chrisdutz, 2024-08-15:
This should explain hour to use snapshots https://plc4x.apache.org/users/getting-started/using-snapshots.html

luedi128, 2024-08-15:
I just tested again with the current 0.13.0-SNAPSHOT (plc4j-driver-eip-0.13.0-20240814.165428-1.jar).
I still see the same issue with the two function block variables with different name length (one odd one even). 
I can request both variables successful with the change above in the PR.

chrisdutz, 2024-08-15:
OK ... odd that I have no issues reading on my devices, but will have a look tomorrow afternoon (when I have my 4h of PLC4X for this week)

chrisdutz, 2024-08-15:
And thanks for submitting a PR ... that's really highly appreciated ;-)

chrisdutz, 2024-08-16:
Should be closed now after merging the corresponding PR:

closed 2024-08-16 by chrisdutz