apache/plc4x #1603: [Bug]: S7 connection always returns "false" on "isConnected()"

chrisdutz, 2024-05-17:
### What happened?

In S7ProtocolLogic the implementation of "isConnected" is as follows:

```
    private boolean isConnected() {
        return context.getChannel().attr(S7HMuxImpl.IS_CONNECTED).get();
        //return true;
    }
```

However the IS_CONNECTED constant never seems to be set, therefore the connection never appears to be connected.

### Version

0.13.0-SNAPSHOT

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2024-05-17:
Actually this is not true ... it's actually not returning false all the time ...

closed 2024-05-17 by chrisdutz

glcj, 2024-05-17:
This access must return the status of the active connection (of the two available).

Anyway, I'm going to check the execution sequence this weekend when I have access to the equipment.