apache/plc4x #1604: [Bug]: modbus rtu maybe error by serial port.

liudianwei, 2024-05-19:
### What happened?

modbus rtu over serial port will not work correct because the serial may be not write all data to the pc(or memory?), due to this, if read serial port immediately after send command, the data just read may be incomplete, so modbus driver will throw some errors like crc or read error.

temporary solution:
[image]
file location plc4x\plc4j\transports\serial\src\main\java\org\apache\plc4x\java\transport\serial\SerialChannelHandler.java
thx.

### Version

latest

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-05-24:
The main problem here is that modbus doesn't have any form of "length indicator" in the protocol.

One option would be to call the parse method in the ByteLengthEstimator in ModbusAsciiDriver and ModbusRtuDriver ... this would increase the cpu and memory-load of the drivers, but serial ports not really being high-speed, I think this doesn't matter much.

chrisdutz, 2024-05-24:
I've just pushed an update that should make Modbus ASCII and RTU a bit more stable in situations as you describe. Please give the SNAPSHOT a try as soon as Jenkins is finished building it.

https://ci-builds.apache.org/job/PLC4X/job/PLC4X/job/develop/2050/

chrisdutz, 2024-08-02:
Reading this issue again, I do think that this issue should be resolved by the addition of the ByteLengthEstimator that tries to parse the message first and uses the size of that message to actually read the message.

Please re-check and confirm this issue is either solved or still present, otherwise I will close it the next time I work on modbus issues.

liudianwei, 2024-08-08:
works fine, thx

closed 2024-08-08 by liudianwei