apache/plc4x #1610: [Bug]:  modbus over serial port(com) could not closed successfully.

liudianwei, 2024-05-23:
### What happened?

modbus over serial port(com) could not closed successfully, try to call PlcConnection.close() when com disconnect, it will be in an endless loop. the major reason is that it will call channel close by netty automatically when shutdown NioEventLoop, and serial channel will put an doClose in eventLoop, it will call close again because eventLoopUp is not empty, and than, trapped in a dead cycle...
temp solution:
close method should only support call once.
[image]
[image]
[image]
file location plc4x\plc4j\transports\serial\src\main\java\org\apache\plc4x\java\transport\serial

### Version

latest

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

liudianwei, 2024-05-23:
link #1342

chrisdutz, 2024-08-02:
As one of the issues you referenced is closed as "works now" could you please re-test if the problem still exists?

liudianwei, 2024-08-08:
still exists.

oliviercoma, 2024-10-24:
We're also impacted by this issue.
I wanted to work on a fix, but the SerialChannel implementation seems really hacky without much information / comments and thus really difficult to work on:
- Why close method from AbstractChannel.AbstractUnsafe is overriden and completely by-passed as it deals with the specific issue (variable closeInitiated)
- Why unsafe() returns a new instance each time (a new instance is recreated at each read). Other implementations do not do that and only newUnsafe() is overriden. As newUnsafe() is not overriden, there is a rogue NioByteUnsafe that lies around.
- Why SerialNioUnsafe.read() implementation is very similar (but a bit different) to the one from NioByteUnsafe. Why is it not overriden?