apache/plc4x #1614: [Bug]: SlaveID

ZantsuRocks, 2024-05-24:
### What happened?

I think that something is not working with the TAG Creation...

I cant build a tag to read data from slave-id 2 in modbus.

I have a network with 10 slave, i need to read them in order but the string won't build

When i remove the "{unit-id 2}" from the tag it works.

```java
//connection
plcConn = plcDM.getConnectionManager().getConnection("modbus-rtu:tcp://127.0.0.1:502"); //FIXME: Somente para debug

// ...Doing things...

builder.addTagAddress("holding-registers", "holding-register:1[23]{unit-id: 2}");

PlcReadRequest readRequest = builder.build();

PlcReadResponse response = readRequest.execute().get();
```

the error:

```
Exception in thread "Thread-2" org.apache.plc4x.java.api.exceptions.PlcInvalidTagException: holding-register:1{unit-id: 2}[23] invalid
	at org.apache.plc4x.java.modbus.base.tag.ModbusTagHandler.parseTag(ModbusTagHandler.java:41)
	at org.apache.plc4x.java.spi.messages.DefaultPlcReadRequest$Builder.lambda$0(DefaultPlcReadRequest.java:122)
	at org.apache.plc4x.java.spi.messages.DefaultPlcReadRequest$Builder.lambda$2(DefaultPlcReadRequest.java:138)
	at java.base/java.util.LinkedHashMap.forEach(LinkedHashMap.java:986)
	at org.apache.plc4x.java.spi.messages.DefaultPlcReadRequest$Builder.build(DefaultPlcReadRequest.java:138)
```

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

StrawberryBlue, 2024-05-28:
slaveId use url parameter as "modbus-tcp:tcp://127.0.0.1:502?request-timeout=5000&default-unit-identifier=1"

ZantsuRocks, 2024-05-28:
For TCP connections there is no problem in this.
But for Serial connections this breaks all the utility of a Slave ID, we can have at least 247 slaves in serial communication, using this parameter as connection parameter just make the use difficult, we need to keep opening and closing the serial connection.
I was using the TCP transport just to debug without a real hardware, my final transport will be serial.

StrawberryBlue, 2024-05-29:
Try this one.
I can succeed.
`builder.addTagAddress("value-1", "holding-register:1:UINT[10]{unit-id: 2}");`

chrisdutz, 2024-06-21:
Can we close this issue?

ZantsuRocks, 2024-06-27:
> Can we close this issue?

I gave up on using this library since my project is heavily dependent on SlaveID. You can close the issue, and if I use the library again, I will check and open another one if necessary.

closed 2024-06-27 by ZantsuRocks