apache/plc4x #1615: [Feature Request]: Hello, how can I communicate with OMRON PLC?

rooterWzs, 2024-05-25:
### What would you like to happen?

Thank you!

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-05-25:
It seems someone would need to implement the omron Fins protocol for that.

Considering, that I dint have any devices, software or access to the specs, this would be challenging. Perhaps someone else from the community can help?

rooterWzs, 2024-05-27:
Then I'll give it a try！

chrisdutz, 2024-05-27:
If you need any help with that (as long as your implementation will be coming back to the project) ... let us know, happy to help mentoring.

hutcheb, 2024-12-23:
Would love to see a FINS driver being developed, but closing due to inactivity. Reach out if you need some help with it.

closed 2024-12-23 by hutcheb