apache/plc4x #1617: [Feature Request]: Can customize ‘default-payload-byte-order’ order

StrawberryBlue, 2024-05-28:
### What would you like to happen?

now,Allowed values are:
- BIG_ENDIAN
- LITTLE_ENDIAN
it is like
- 3412
- 1234
- 4321
- 2143

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-05-28:
Actually the versions I implemented are more like this:

BIG_ENDIAN = 1234
LITTLE_ENDIAN = 4321

The other two options: 3412 and 2143 we didn't need at that time and couldn't find devices that need this.
We would probably call them something else.

However implementing these options is more work than the first two. I personally would probably not be working on this any time soon. If you want me to prioritize this, I have this page, wehre you can drop a donation: https://buymeacoffee.com/christoferu which will definitely motivate me to do it sooner .... or - what would be even better, you whip up a PR for this feature yourself or you find someone who does it for you.

StrawberryBlue, 2024-05-29:
Okay, I'll try to implement it

chrisdutz, 2024-05-29:
Naming wise, I would opt for:
- BIG_ENDIAN_WORD_SWAP 
- LITTLE_ENDIAN_WORD_SWAP

chrisdutz, 2024-06-21:
Ok ... today I implemented the feature of supporting BIG_ENDIAN_BYTE_SWAP and LITTLE_ENDIAN_BYTE_SWAP as well as added support for overriding the default setting in the connection string on a per-tag basis.

closed 2024-06-21 by chrisdutz