apache/plc4x #1625: [Bug]: Issue with PLC4J BACnet Driver - Active Connections Not Supported

zhucPPl, 2024-05-30:
### What happened?

Content:
I have encountered an issue while using the plc4j-driver-bacnet (version 0.12.0). Upon creating a connection, I received the following error message: "Active connections not yet supported." Upon inspecting the source code, I discovered that the issue stems from the protocol being set to passive=false. Further investigation revealed that none of the supported transports (udp, tcp) are returning true, indicating that they do not support active connections. Additionally, when attempting to use the 'pcap' transport, I received the error message "Unsupported transport pcap," which has left me confused.

Here is the relevant code snippet:

java
String connectionString = "bacnet-ip:pcap://127.0.0.1:47808";
PlcConnection plcConnection = PlcDriverManager.getDefault().getConnectionManager().getConnection(connectionString);

I can confirm that the address used in the connection string is valid, as I have successfully implemented BACnet using bacnet4j. Currently, I have utilized plc4j to implement the modbus and knx protocols without issues, but encountered this problem while implementing the BACnet protocol. I would greatly appreciate any assistance in resolving this matter. Thank you.

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [X] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-05-30:
This is actually not a bug.
Active communication in Java has never been implemented.
The beginning of the bacnet driver was a paid gig, where I had to build a driver to passively observe a bacnet network.

Sebastian worked hard on an active version in go. Not quite sure about the state there.

Right now I have no plans to do the implementation of an active bacnet driver in Java.

Perhaps others can help here or you want to get your fingers dirty :-)

I think the heavy lifting of defining all the types is done and finished.

Chris

zhucPPl, 2024-05-30:
Thank you for your answer and for your hard work

sruehl, 2024-05-31:
BACnet in golang is not usable yet