apache/plc4x #1662: [Bug]:

stegal-bh, 2024-06-13:
### What happened?

I wanted to run plc4x-opcua-server jar file, but I'm getting the following error:

```
SLF4J(W): No SLF4J providers were found.
SLF4J(W): Defaulting to no-operation (NOP) logger implementation
SLF4J(W): See https://www.slf4j.org/codes.html#noProviders for further details.
```
JDK version: "17.0.9" 2023-10-17 LTS
Maven version: Apache Maven 3.9.7

My steps performed:

1. Download release v0.12.0 version [link to v0.12.0](https://github.com/apache/plc4x/tree/v0.12.0)
2. Go to the **opcua-server** folder (_plc4j\tools\opcua-server_)
3. Perform `mvn package` command
4. Go to the **plc4x-opcua-server-0.12.0-uber-jar.jar** package at _plc4j\tools\opcua-server\target_ folder
5. Perform `java -jar plc4x-opcua-server-0.12.0-uber-jar.jar -c config.yml -i` command
6. I'm getting error
[image]

### Version

v.0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

hutcheb, 2024-06-13:
It looks as though no logger class files were found in your class path. (I spent a few minutes trying to add one on my machine without success). I would expect if you manually added a logger class to your command it should find it.

Not sure if you mentioned if it exits afterwards or not. If it does you might need to create the config file, this should be based on the sample config.yml in the main/resources directory (make sure to uncomment the security directory you want to use). Otherwise fixing the logging issue would give you more information.

stegal-bh, 2024-06-13:
Hi @hutcheb, I have found the problem. 
I added **slf4j-simple** as dependency and 

```
  <dependency>
    <groupId>org.slf4j</groupId>
    <artifactId>slf4j-simple</artifactId>
    <version>2.0.13</version>
  </dependency>
```
and uncommented `dir: "./security"` in yml config file and now it's working fine.

thanks

ottlukas, 2024-06-15:
> Hi @hutcheb, I have found the problem. I added **slf4j-simple** as dependency
> 
> ```
>   <dependency>
>     <groupId>org.slf4j</groupId>
>     <artifactId>slf4j-simple</artifactId>
>     <version>2.0.13</version>
>   </dependency>
> ```
> 
> and uncommented `dir: "./security"` in yml config file and now it's working fine.
> 
> thanks

Does that mean this issue can be closed?

chrisdutz, 2024-06-16:
Well I think it hasn#t been changed in the code ... so I would say: nope.

chrisdutz, 2024-06-28:
When running the opcua-server in SNAPSHOT, I'm getting this message:

```
cdutz@MacBook-Pro-von-Christofer target % java -jar plc4x-opcua-server-0.13.0-SNAPSHOT.jar -c config.yml -i
Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.SecurityException: Invalid signature file digest for Manifest main attributes
```
Could someone from the OPC-UA folks please have a look at it?
@hutcheb @splatch ?

chrisdutz, 2024-06-30:
Ok ... so I had a look at the OPC-UA server in it's current state and it was definitely missing some configuration to merge all driver configs. I was only able to start it with one driver available, as the shade plugin was overwriting the META-INF/services/ property files instead of merging them ... after adding that transformer, I was able to start the OPC-UA server without errors and without the logging message. I think this issue can be closed ... do you agree?

closed 2024-07-05 by chrisdutz