apache/plc4x #1670: [Bug]: ModbusTag - IllegalArgumentException of "quantity may not be larger than"

yangshuyi, 2024-06-19:
### What happened?

When I try to using batch mode to query out all binary data (more than 2000) by specify a StartAddress and a length = 3000, it will throw exception of IllegalArgumentException.

Currently I can only split all binary data into small group. But it is not nice for me.

Can anyone help to clarify why p4j modbus module need this limitation? Is there any other workaround way or configuration? 

Thanks a lot.

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-06-19:
I think this is not a limitation of the plc4x driver, but if the protocol or the device you are taking to. Every type of plc has a maximum packet size (except Ads). So at sind point in time you need to split up the request. For some protocols (such as s7) this is well defined and we automatically split up behind the scenes. For modbus this "optimizer" is not yet built.

yangshuyi, 2024-06-19:
Understood, I check with the modbus protocol. It has such limit.

Thanks for your kindly answer.

closed 2024-06-19 by yangshuyi

chrisdutz, 2024-06-19:
So you have a reference that documents this maximum size? If it's official, we could probably build a general purpose optimizer.

yangshuyi, 2024-06-19:
Please refer to: https://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf, page: 11

[image]

chrisdutz, 2024-06-19:
Aaaaah ... yeah ... ok ... so they don't directly define a max PDU Size like other protocols, but limit the "instance" number. But I still think we could build an Optimizer for that. If you want, you can have a look at the Siemens S7 QueryOptimizer ... you could use this as a template to rewrite PLC4X api calls and transparrently rewrite them to series of requests. This way you don't need to worry about these protocol limitations. I'm happy to be your mentor for your first PLC4X contributions, if you like. If you don't want to or are not able to, this feature will take quite a bit of time as I no longer get the time allowance to work on PLC4X which I used to have.

chrisdutz, 2024-06-19:
Created a Feature Request here: #1671