apache/plc4x #1671: [Feature Request]: Build a QueryOptimizer for PLC4J Modbus

chrisdutz, 2024-06-19:
### What would you like to happen?

The modbus spec seems to provide limits for the size of packets. However in contrast to the other protocols, is this limit not related to the size of the packet, but by the number of registers/coils. 

Please refer to: https://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf, page: 11

It should be simple to implement a query-optimizer for Modbus to automatically handle this. The Siemens S7 QueryOptimizer might help with inspiration.

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-06-19:
So it seems that the limitations are:

Max 2000 coils in one request (Page 11)
Max 125 registers in one request (Page 15)

chrisdutz, 2024-09-04:
So luckily I need to implement this feature for my $job ;-) Making good progress, however I'm only going to implement this for reading ... writing would possibly open pandora's box with evil issues, which I do not want to do.

chrisdutz, 2024-09-04:
[image]
So this is the first rough draft ... now I gotta add a bit more of testing to everything ... but more than 10 times faster reads is quite an optimization ;-)

glcj, 2024-09-04:
Excellent work, I will run tests, those times are excellent!

chrisdutz, 2024-09-05:
Ok ... the changes are merged ... now I think it makes sense to wait another week for some feedback.

closed 2024-10-18 by chrisdutz