apache/plc4x #1675: [Bug]: CIP service is not supported When connect to Allen Bradley PLC

acfly, 2024-06-20:
### What happened?

I need to read multiple tags from ControlLogix 1756-L71, for example 1000 tags,  and I found it quite slow, so i debug into the driver source code,  and it appears that the 247th line in `EipProtocolLogic.java` always trigged.
```
if ((long) response.getStatus() == CIPStatus.ServiceNotSupported.getValue()) {
                    context.fireConnected();   # this 247th line
                    return;
                } else if ((long) response.getStatus() != CIPStatus.Success.getValue()) {
                    throw new PlcRuntimeException("Got status code while polling for supported CIP attributes [" + response.getStatus() + "]");
                }
                if (response.getAttributes() != null) {
                    for (Integer classId : response.getAttributes().getClassId()) {
                        if (CIPClassID.enumForValue(classId) == CIPClassID.MessageRouter) {
                            this.useMessageRouter = true;     // never run
                        }
                        if (CIPClassID.enumForValue(classId) == CIPClassID.ConnectionManager) {
                            this.useConnectionManager = true;   // nerver run
                        }
                    }
                }
```
which means the CIP service is not supported,  and the 665th line in the same file will always be trigged,
```
    public CompletableFuture<PlcReadResponse> read(PlcReadRequest readRequest) {
        CompletableFuture<PlcReadResponse> future;
        if (!this.useMessageRouter && !this.useConnectionManager) {
            future = readWithoutMessageRouter(readRequest);       # 665th line,  read tag one by one 
        } else if (this.useMessageRouter && !this.useConnectionManager) {
            future = readWithoutConnectionManager(readRequest);  # never run, read tags at one request, high efficient
        } else {
            future = readWithConnectionManager(readRequest);     # never run,  read tags at one request, high efficient
        }
        return future;
    }
```
the method `readWithoutMessageRouter` read tags one by one, which cause significant decrease in efficiency. the method `readWithoutConnectionManager` and `readWithConnectionManager` are reading tags at one request, but they will never be triggered.

there is a similar bug reported and the author give a solution. [Unable to Read from Allen-Bradley 1756-L73 A ControlLogix PLC](https://github.com/libplctag/libplctag/issues/437#issue-1904212613)  

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-06-25:
I'll look into this, however don't expect anything to happen soon ... I only have 4h per week to work on PLC4X and there's a pretty long list of other things that also need to be fixed. If you want to whip up a PR, I will be more than happy to have a look at this.

chrisdutz, 2024-06-27:
So I can confirm, that most probably the problem is that the current driver uses the new larger forward open message, which only newer Logix models seem to support. So I guess the probably simplest solution would be adding a config option to the protocol, telling the driver to use the older forward_open messages (which might also require automatic splitting of requests into multiple messages if too many are requested)

This is a feature, that I will most probably not be able to tackle with the 4h per week that I have. If I would this would be consuming all of my available time for many weeks. Perhaps someone here will be able to help.

chrisdutz, 2024-07-12:
I just pushed a change, that allows adding a "force-unconnected-operation" connection string parameter to make the driver use the unconnected option. Also did I apply some other changes to that method that should make it more stable.

chrisdutz, 2024-07-12:
Please test.

chrisdutz, 2024-07-26:
If I don't hear back from you till next time I work on this (Probably next Friday) I'll mark this issue as done and close it.

chrisdutz, 2024-08-16:
Closing ... issue seems to be addressed and reporter didn't say anything otherwise.

closed 2024-08-16 by chrisdutz