apache/plc4x #1702: [Bug]: Some point readings in EthernetIP data read are null. In EipProtocolLogic, the decodeSingleReadResponse method returns a result where CipService p is not a CipReadResponse.

zhang13, 2024-07-13:
### What happened?

Some point readings in EthernetIP data read are null. In EipProtocolLogic, the decodeSingleReadResponse method returns a result where CipService p is not a CipReadResponse. p is CipConnectedResponse.
Only certain fixed points return p as CipConnectedResponse, and I don't know why.
`private Map<String, ResponseItem<PlcValue>> decodeSingleReadResponse(CipService p, String tagName, PlcTag tag) {
        Map<String, ResponseItem<PlcValue>> values = new HashMap<>();
        if (p instanceof CipReadResponse) {
            CipReadResponse resp = (CipReadResponse) p;
            PlcResponseCode code = decodeResponseCode(resp.getStatus());
            PlcValue plcValue = null;
            CIPDataTypeCode type = resp.getData().getDataType();
            ByteBuf data = Unpooled.wrappedBuffer(resp.getData().getData());
            if (code == PlcResponseCode.OK) {
                plcValue = parsePlcValue((EipTag) tag, data, type);
            }
            ResponseItem<PlcValue> result = new ResponseItem<>(code, plcValue);
            values.put(tagName, result);
        } else {
// **p is CipConnectedResponse**
            System.out.println(p.getClass().getName()); // CipConnectedResponse
        }
        return values;
    }`

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-07-13:
Would you be able to do a wireshark recording of your communication? Also did I work on the EIP driver yesterday and fixed a few things ... so I would suggest to try the latest version. 

Unfortunately there's something wrong with our build VM, so we are not able to provide new SNAPSHOT versions at the moment, so you would have to checkout and build PLC4X yourself.

zhang13, 2024-07-19:
The following are the packet capture commands. The last one has an issue.

server: 04 00 00 00 00 00 00 00 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 
plc:    04 00 1A 00 00 00 00 00 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 01 00 00 01 14 00 01 00 20 01 43 6F 6D 6D 75 6E 69 63 61 74 69 6F 6E 73 00 00 

server: 65 00 04 00 00 00 00 00 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 01 00 00 00 
plc:    65 00 04 00 00 38 02 13 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 01 00 00 00 

server: 6F 00 16 00 00 38 02 13 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 B2 00 06 00 01 02 20 02 24 01 
plc:    6F 00 14 00 00 38 02 13 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 B2 00 04 00 81 00 08 00 

server: 6F 00 38 00 00 38 02 13 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 B2 00 28 00 52 02 20 06 24 01 05 9D 1A 00 4C 0B 91 0A 54 55 52 4E 53 49 47 4E 41 4C 91 07 43 4F 4E 54 52 4F 4C 00 01 00 01 00 01 00 
plc:    6F 00 1A 00 00 38 02 13 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 B2 00 0A 00 CC 00 00 00 C4 00 0C 00 00 00 

server: 6F 00 3F 00 00 38 02 13 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 B2 00 2F 00 52 02 20 06 24 01 05 9D 21 00 4C 0E 91 0B 4C 41 4E 45 53 49 47 4E 41 4C 31 00 91 0C 49 42 41 43 4B 58 53 54 41 54 55 53 00 01 00 01 00 01 00 
plc:    6F 00 18 00 00 38 02 13 00 00 00 00 50 4C 43 34 58 20 20 20 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 B2 00 08 00 D2 00 01 01 05 02 00 00

chrisdutz, 2024-07-26:
Ok ... so I found out .... your system is "little-endian" ... and indeed something is going wrong on the last interaction (both request and response)

chrisdutz, 2024-07-26:
So from your dumps I could see, that it seems the driver does send out a CipConnectedResponse and gets a CipConnectedResponse back, which is odd. What type of PLC are you using and what's the setting in your connection-string?

zhang13, 2024-07-28:
Connection Configuration: eip:tcp://127.0.0.1:44818?backplane=1&slot=0&bigEndian=false
PLC Model: Allen-Bradley (AB) 1756-A17
Issue Description: Most point readings are correct, but some specific points have issues. For example:
LANESIGNAL1.IFRONTXSTATUS:BOOL is correct.
LANESIGNAL1.IBACKARROWSTATUS:BOOL is incorrect.
Could this be related to the point configuration or certain settings?

Of course, I made modifications to the EipTag validation. I'm not sure if this affects the issue because the provided tags do not match the existing pattern.

`
private static final Pattern ADDRESS_PATTERN = Pattern.compile("^(?<tag>[%a-zA-Z_.0-9]+\\[?[0-9]*]?[%a-zA-Z_.0-9]+):?(?<dataType>[A-Z]*):?(?<elementNb>[0-9]*)");
`

DC2-DanielKrueger, 2024-07-30:
We face a very similar issue, within the code a `CipConnectedResponse` is found while a `CipReadResponse`is expected:

```
io.netty.handler.codec.DecoderException: java.lang.ClassCastException: class org.apache.plc4x.java.eip.readwrite.CipConnectedResponse cannot be cast to class org.apache.plc4x.java.eip.readwrite.CipReadResponse (org.apache.plc4x.java.eip.readwrite.CipConnectedResponse and org.apache.plc4x.java.eip.readwrite.CipReadResponse are in unnamed module of loader com.hivemq.edge.modules.adapters.impl.IsolatedModuleClassloader @476f723d)
```

We actually use `org.apache.plc4x:plc4j-driver-eip:0.12.0`but the stacktrace is from `0.11.0` as interestingly a previous state of our software worked. However just decrementing the dependency version did not do the trick. 

```
17:20:57.686 [nioEventLoopGroup-2-1] DEBUG org.apache.plc4x.java.spi.GeneratedDriverByteToMessageCodec - Sending bytes to PLC for message 
<graphical output emitted> 
as data 6f0030000030020d00000000504c43345820202000000000000000000000020000000000b2002000520220062401059d12004c07910b6465765f696e745f74616700010001000000
```

this seems to be the received data for the request:

```
17:20:57.729 [nioEventLoopGroup-2-1] DEBUG org.apache.plc4x.java.spi.generation.io.MyDefaultBitInput - Before
00|6f 00 18 00 00 30 02 0d 00 00 'o....0....'
10|00 00 50 4c 43 34 58 20 20 20 '..PLC4X   '
20|00 00 00 00 00 00 00 00 00 00 '..........'
30|02 00 00 00 00 00 b2 00 08 00 '..........'
40|d2 00 01 01 11 03 01 00       '........  '
17:20:57.729 [nioEventLoopGroup-2-1] DEBUG org.apache.plc4x.java.spi.generation.io.MyDefaultBitInput - aligned 0
17:20:57.729 [nioEventLoopGroup-2-1] DEBUG org.apache.plc4x.java.spi.generation.io.MyDefaultBitInput - set to index 44
17:20:57.729 [nioEventLoopGroup-2-1] DEBUG org.apache.plc4x.java.spi.generation.io.MyDefaultBitInput - After
00|6f 00 18 00 00 30 02 0d 00 00 'o....0....'
10|00 00 50 4c 43 34 58 20 20 20 '..PLC4X   '
20|00 00 00 00 00 00 00 00 00 00 '..........'
30|02 00 00 00 00 00 b2 00 08 00 '..........'
40|d2 00 01 01 11 03 01 00       '........  '
```

The library is also plotting the packets. I left it out as it is hard to read on github.

chrisdutz, 2024-08-01:
Unfortunately I haven't managed to get any of my other PLCs working with EIP, so it's a bit challenging for me to test. Are there any emulators available that i could use to test?

DC2-DanielKrueger, 2024-08-01:
0.13.0-Snapshot works again. Only 0.12.0 does not work currently. Is there a known regression between 0.11.0 and 0.12.0? I will try to figure out what is going wrong.

DC2-DanielKrueger, 2024-08-01:
@zhang13 I guess you need to add the little endian to your connection string. You may take a look at EIPProtocolAdapter:createQueryStringParams (https://github.com/hivemq/hivemq-edge/blob/master/modules/hivemq-edge-module-plc4x/src/main/java/com/hivemq/edge/adapters/plc4x/types/eip/EIPProtocolAdapter.java). We set "bigEndian" to false here. One caveat: 0.13.0 changes the key to "big-endian".

chrisdutz, 2024-08-02:
Did you do anything to fix it in 0.13.0-SNAPSHOT? Otherwise I have spent my Weekly PLC4X-Time on EIP for several weeks now, trying to improve things. Perhaps one of these changes fixed things?

DC2-DanielKrueger, 2024-08-02:
No I did no changes to 0.13.0.  So this what it looks like: 
0.11.0 => works
0.12.0 => does not work
0.13.0-SNAPSHOT => works again

chrisdutz, 2024-08-02:
Ok ... so I guess some of the changes I did had a positive effect ;-) That sounds prommissing ;-)

chrisdutz, 2024-08-02:
And the main reason why things changed with 0.12.0 were that in this version we added a completely rewritten version of the original driver. This now was able to support more complex communication forms and explicitly support Logix devices. I would assume that a performance comparison of 0.11.0 and 0.13.0-SNAPSHOT should possibly show a much better performance, as we're not doing everything in single-address requests as I think it was done till 0.11.0.

chrisdutz, 2024-08-02:
And as we will not be fixing 0.12.0 and the issue seems to be fixed in 0.13.0-SNAPSHOT, can this issue then be closed?

chrisdutz, 2024-08-16:
Closing because it seems the issue is resolved and we got no differeng feedback from the reporter.

closed 2024-08-16 by chrisdutz