apache/plc4x #1705: [Bug]: OPCUA server and OPCUA failed to read data

kdxq, 2024-07-17:
### What happened?

uaExpert can read data,just like follow;
[image]
I use opcua-server,config as follow:
[image]
and use opcua to read data from opcua-server, code as follow
[image]
but the FanSpeed return Null
[image]
what's wrong? please give me some suggest,thanck you very much!

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

kdxq, 2024-07-17:
I think the reson is in the opcua-server.
I use 0.11.0, the opcua-server throw error as follow:
[image]
0.12.0, the opcua-server throw error as follow:
[image]

splatch, 2024-07-17:
The incompatible datatype exception means that value you expect to be a float is encoded differently. Try first switching `getFloat` to `getObject` to determine how it is being transmitted. Second step - with 0.12 you can also specify read value type in opc ua tag address.

kdxq, 2024-07-17:
I used plc4x-opcua-server  as server, and used ua_export as client.
when I add the url to ua_export client pop error as follow:
[image]
then, I ignore it, ua_export show the data structure as follow:
[image]
and get data in loop, error as follow
[image]

kdxq, 2024-07-17:
[image]

0.11.0 version，eip part got values ， but opcua part still error。

splatch, 2024-07-17:
@kdxq Can you try `ns=1;s=FanSpeed;REAL` syntax for OPC UA tags?

kdxq, 2024-07-17:
@splatch  do you mean like this? still failed
[image]

kdxq, 2024-07-17:
by the way. when I write data. only float type is ok, all others type all failed.as follow

[image]

[image]

kdxq, 2024-07-18:
@splatch   did you mean like this? still failed
[image]

kdxq, 2024-07-23:
Hi，@ottlukas ，@splatch，I have tested the opcu ua server from 0.10.0 to 0.13.0, but none of them were able to read data, and none of the data types worked. Do you have any other testing suggestions? Additionally, the testing sources for the OPCUA I am using are publicly available online. Can you provide some guidance on how to conduct the testing? I want to make plc4x better and more stable.

splatch, 2024-07-23:
@kdxq I’m going to work on some integration tests which will need working opc ua server. I’ll try to pull this issue as part of test setup.
I’ll work on plc4j from 1st August.

splatch, 2024-08-08:
@kdxq I've had a look on behavior of starfive demo server and there is something odd in there.
I could not fully reproduce your issue. To be fair, I could not connect to demo server at all with 0.13-SNAPSHOT. Interestingly, local instance of node-opcua do not have this issue.

I've found that with 0.13 anonymous authentication differs between clients. We do not supply client nonce nor certificate for create session request if security mode is NONE. I'll try to fix connectivity issue first and then see if readout issue still remains.

qtvbwfn, 2024-08-10:
I connected to your service and, based on the tool's view, the data type should be Double, not Float.

[image]

qtvbwfn, 2024-08-10:
@kdxq 
I connected to your service and, based on the tool's view, the data type should be Double, not Float.

[image]

closed 2024-10-17 by splatch

splatch, 2024-10-17:
@kdxq I've pushed a fix which fixed compatibility issue with demo server from starfive / node-opcua. Please test current develop and re-open this issue in case if bug you spotted still remains.

The timeout errors you was getting was likely caused by client not really connecting to demo server, and so making it impossible to fetch tag value.

commit 716e280a (2024-10-17): fix(plc4j): Adjust endpoint selection logic.

Amended selection logic made use of streams with side effects.
When server had more than one endpoint and these endpoints had same policy (i.e. anonymous) with different identifiers (i.e. anonymous_1, anonymous_2), it lead to wrong activate session requests which mixed endpoint and user token policy.
Updated logic removes side effects and makes sure that only policies of matched server endpoint are tested against client configuration.

Closes #1705.
commit a159e7ed (2024-10-17): fix(plc4j): Adjust endpoint selection logic.

Amended selection logic made use of streams with side effects.
When server had more than one endpoint and these endpoints had same policy (i.e. anonymous) with different identifiers (i.e. anonymous_1, anonymous_2), it lead to wrong activate session requests which mixed endpoint and user token policy.
Updated logic removes side effects and makes sure that only policies of matched server endpoint are tested against client configuration.

Closes #1705.
PR #1830 (2024-10-17): fix(plc4j): Adjust endpoint selection logic.
Amended selection logic made use of streams with side effects. When server had more than one endpoint and these endpoints had same policy (i.e. anonymous) with different identifiers (i.e. anonymous_1, anonymous_2), it lead to wrong activate session requests which mixed endpoint and user token policy. Updated logic removes side effects and makes sure that only policies of matched server endpoint are tested against client configuration.

Closes #1705.