apache/plc4x #1711: Is there any plan time for C

qljsky, 2024-07-29:
### What would you like to happen?

hi,Is there any plan time  about plc4c

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [X] plc4c
- [ ] plc4net

### Protocols

- [X] AB-Ethernet
- [X] ADS /AMS
- [X] BACnet/IP
- [X] CANopen
- [X] DeltaV
- [X] DF1
- [X] EtherNet/IP
- [X] Firmata
- [X] KNXnet/IP
- [X] Modbus
- [X] OPC-UA
- [X] S7

chrisdutz, 2024-07-29:
There is no budget to help develop, so there is no time-plan at all. Everything will happen if somone needs it and is willing to invest the time to do it, or finance someone else to do it for him.

chrisdutz, 2024-08-16:
So if there is no feeback on either options (willing to work on it yourself or that there's budget availalbe) I think I will close this issue with the answer "No" next time I come here ;-)

closed 2024-08-30 by chrisdutz