apache/plc4x #1748: [Bug]: Writing into tag of type REAL gives java.nio.BufferOverflowException

fdupont-noovelia, 2024-09-06:
### What happened?

Writing into a tag of type REAL gives « java.nio.BufferOverflowException ».

If I am not mistaken, « putDouble » should be « Float »?

Connection String Options :
eip://  :44818?backplane=1&slot=0&big-endian=false&force-unconnected-operation=true

Call stack
```json
{
    "throwable": {
        "className": "java.nio.BufferOverflowException",
        "message": "null",
        "stepArray": [{
                "className": "java.nio.Buffer",
                "methodName": "nextPutIndex",
                "fileName": "Buffer.java",
                "lineNumber": 730
            }, {
                "className": "java.nio.HeapByteBuffer",
                "methodName": "putDouble",
                "fileName": "HeapByteBuffer.java",
                "lineNumber": 609
            }, {
                "className": "org.apache.plc4x.java.eip.base.protocol.EipProtocolLogic",
                "methodName": "encodeValue",
                "fileName": "EipProtocolLogic.java",
                "lineNumber": 1305
            }, {
                "className": "org.apache.plc4x.java.eip.base.protocol.EipProtocolLogic",
                "methodName": "writeWithoutMessageRouter",
                "fileName": "EipProtocolLogic.java",
                "lineNumber": 926
            }, {
                "className": "org.apache.plc4x.java.eip.base.protocol.EipProtocolLogic",
                "methodName": "write",
                "fileName": "EipProtocolLogic.java",
                "lineNumber": 1228
            }, {
                "className": "org.apache.plc4x.java.spi.connection.AbstractPlcConnection",
                "methodName": "write",
                "fileName": "AbstractPlcConnection.java",
                "lineNumber": 199
            }, {
                "className": "org.apache.plc4x.java.spi.messages.DefaultPlcWriteRequest",
                "methodName": "execute",
                "fileName": "DefaultPlcWriteRequest.java",
                "lineNumber": 58
            }
        ]
    }
}
```

### Version

0.13.0-SNAPSHOT

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

fdupont-noovelia, 2024-09-06:
.close-issue

fdupont-noovelia, 2024-09-06:
Problem seems on my side.

fdupont-noovelia, 2024-09-09:
Changing 

buffer.putDouble(value.getDouble());
to
buffer.putFloat(value.getFloat());
in
plc4j\drivers\eip\src\main\java\org\apache\plc4x\java\eip\base\protocol\EipProtocolLogic.java

Solves the issue.

Our program is in kotlin. It appears to be more sensitive to data types.

fdupont-noovelia, 2024-09-09:
.add-labels java,Ethernet/IP

ottlukas, 2024-09-10:
Can this issue then considered as solved and it can be closed again?

chrisdutz, 2024-09-10:
Well not if the change has not been included in EipProtocolLogic.java

chrisdutz, 2024-09-10:
So .. now it's in ... please test

closed 2024-09-10 by chrisdutz

fdupont-noovelia, 2024-09-10:
Yes, now I can successfully write into REAL tag with our Kotlin program. Thanks

I see in the commit LREAL was also added, I don't have anything on hand to test that one.

chrisdutz, 2024-09-10:
Yeah ...  well the original code was handling LREAL data under the name of REAL, so I took the liberty of implementing both.

commit ba5fcded (2024-09-10): fix: Updated the handling of real values in EIP

possibly fixes:  #1748