apache/plc4x #1754: [Bug]: ADS, reading array returns only 1st element

fdupont-noovelia, 2024-09-09:
### What happened?

PlcReadResponse getAllBytes/getAllIntegers with PLC4J to a Beckhoff TwinCat program :

arrayVariable AT %I* : ARRAY [1..13] OF WORD;

Returns only the 1st element.

Array has :
[11, 22, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12, 13]

[11] is returned.

New issue since commit : bdd05c9e765bfca294b20b398f346b241a2e0aa3 [bdd05c9]
Date: September 8, 2024 13:14:19

### Version

0.13.0-SNAPSHOT

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [X] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-09-09:
The code changed in that commit only affected the browse functionality ... not sure how this should have affected the read-functionaility.

chrisdutz, 2024-09-09:
Oh ... having a deeper look at it ... it does seem as if the changes to getPlcValueTypeForAdsDataType really do have an effect on the reading of arrays ... but I'm currently cleaning up the array handling as currently it wouldn't be possible to address an array element where the array part it not at the end. So consider me working on it.

But thinking about it ... I guess I could do a quick-fix for it.

chrisdutz, 2024-09-09:
I whish Github would not automatically close things immediately when you commit a potential fix.

closed 2024-09-09 by chrisdutz

fdupont-noovelia, 2024-09-09:
With the last commit, reading array works again. Thanks.

commit 1d8c7e06 (2024-09-09): fix: Split up the datatype guessing function into one used for browse and one used for reading

fixes:  #1754