apache/plc4x #1764: [Flaky-Test]: A test in the java OPC-UA driver is continuously breaking builds.

chrisdutz, 2024-09-16:
### What happened?

The test OpcuaPlcDriverTest.multipleThreads in the java opc-ua driver testsuite is regularly failing the CI/CD builds ... re-runing them usually resolves the issue, however we can't have test break the builds at random.

### Version

v0.13.0-SNAPSHOT

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

chrisdutz, 2024-09-16:
.add-labels flaky-test

splatch, 2024-09-16:
I'll take a look on these and the end of week or early next. I hoped that testcontainers would help us a bit with this issue, but there still seem to be a problem.

chrisdutz, 2024-09-16:
Admittedly I think the issue lies with milo and it therefore doesn't really matter how you start it ... but admittedly I didn't look too deep into the issue.

chrisdutz, 2024-09-30:
org.apache.plc4x.java.opcua.protocol.chunk.ChunkFactoryTest also seems to continuously break builds (all values seem to be off by 1)

splatch, 2024-10-22:
For now I've changed assertion in `ChunkTest` to see if we have this +1 offset only for one field or for many. I am not entirely sure why it happens. I had a look and there is fairly basic math which is static.

Lets keep this issue open to track this test and others, if they keep failing randomly.