apache/plc4x #1776: [Bug]: Do not throw PlcInvalidTagException exception during batched PlcReadRequest

vladbogun1, 2024-09-20:
### What happened?

[image]
i am performing reading multiple tag addresses at once, and if one of tag not exist on PLC side, whole request will fail with exception. How can i get response for other tag addresses? Or is there any method that can help me filter out from my request invalid tags?

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [X] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-09-20:
Ok ... making progress ...

[image]

This also continues work on the implementation of reading data without previously reading the symbol and data-type table.
However I still need to figure out how to dynamically read data-type information.

chrisdutz, 2024-09-20:
Ok ... please test again with the latest SNAPSHOT version

vladbogun1, 2024-09-23:
Yes, works perfectly, thanks a lot, would it be deployed in some repo as snap version? as i just pull your code and build in local repo by myself
[image]

vladbogun1, 2024-09-23:
and as well i wanna ask you another question about symbolic tags, dont wanna raise bug, etc...

Can we pull using symbolic addresses Array[1] for example? 

[image]
For example i dont wanna make PlcReadRequest for whole `Mod3.FDSCM02.HMI.HW.sInstancePath` that will return me whole array, but for example `Mod3.FDSCM02.HMI.HW.sInstancePath[3]` to get 4rth element. is it real with current implementation or this should be as enchasement?

chrisdutz, 2024-09-23:
Well the SNAPSHOTS are always automatically updated in apache's Nexus ... see here for how to use them:
https://plc4x.apache.org/users/getting-started/using-snapshots.html

And regarding the arrays ... yes ... I will have to put a bit of work into that ... I also noticed that recently. This will most probably require a major refactoring of the driver. Especially if you have arrays of UDTs or even more complex tree and array structures such as:

    Main.someProperty[3][6].someChildProperty[2].yetAnotherChildProperty

Or even worse, when selecting array ranges ... 
Is also not supported ... I am planing on working on this, however it's not going to be done quickly.

chrisdutz, 2024-09-23:
Aaaaaa ... as SNAP (Like Ubuntu snap) ... unfortunately not that I know of ... I thought you were referring to SNAPSHOT.

vladbogun1, 2024-09-23:
> Aaaaaa ... as SNAP (Like Ubuntu snap) ... unfortunately not that I know of ... I thought you were referring to SNAPSHOT.

Actually you was right first time, thats my typo, i mean SNAPSHOT, already implemented your apache repo. thanks a lot for your help)

And you are right about complexity and huge refactoring for Array index feature.
Should i define [Feature Request] followup for arrays?)

chrisdutz, 2024-09-23:
Yeah ... will be working on that anyway as I need that for a private side-project ... but I think in general we really need to improve the array handling in almost every driver we have. Just with drivers supporting symbolic addressing it gets a lot more tricky :-/

closed 2024-09-24 by vladbogun1