apache/plc4x #1780: [Bug]: PLC4X does not wait for response when reading data from Beckhoff Twincat 3 PLC with Ethernet/IP

JoonasC, 2024-09-23:
### What happened?

It seems like PLC4X does not wait for a response from the Ethernet/IP server when reading data (specifically using the Ethernet/IP scanner device in Beckhoff Twincat3 PLCs)

I'm using the following code to test reading:
```java
public class Main {
  public static void main(String[] args) {
    String connectionString = "eip://10.50.100.216:44818?bigEndian=false";

    try (PlcConnection plcConnection = PlcDriverManager.getDefault().getConnectionManager().getConnection(connectionString)) {
      PlcReadRequest.Builder readBuilder = plcConnection.readRequestBuilder();
      readBuilder.addTagAddress("Var1", "%Var1");
      PlcReadRequest readRequest = readBuilder.build();
      System.out.println("--- Reading data ---");
      PlcReadResponse readResponse = readRequest.execute().get();
      System.out.println("--- Read complete ---");
      System.out.println(readResponse.getResponseCode("Var1"));
      System.out.println(readResponse.getObject("Var1"));
    } catch (Exception e) {
      System.out.println("Connection failed");
      e.printStackTrace();
    }
  }
}
```

Stacktrace:

> org.apache.plc4x.java.api.exceptions.PlcInvalidTagException: Var1 invalid
> 	at org.apache.plc4x.java.spi.messages.DefaultPlcReadResponse.getResponseCode(DefaultPlcReadResponse.java:100)
> 	at org.example.Main.main(Main.java:20)

Looking at Wireshark, PLC4X closes the connection before the PLC has a chance to respond to the read request:
[image]

[Capture.pcapng.gz](https://github.com/user-attachments/files/17098250/Capture.pcapng.gz)

I can workaround this bug by adding a manual `Thread.sleep` before trying to interact with the read response object:
```java
System.out.println("--- Reading data ---");
PlcReadResponse readResponse = readRequest.execute().get();
Thread.sleep(1000);
System.out.println("--- Read complete ---");
```

Here is my Twincat3 Ethernet/IP explicit messaging configuration:
[image]

### Version

0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-09-23:
Just a quick question ... why would you access a Beckhoff ADS device with EIP and not ADS? It's sort of like having a Porsche in the garage but driving with the Dacia ;-)

JoonasC, 2024-09-23:
I'm investigating the Ethernet/IP functionality of PLC4X, but do not have any other Ethernet/IP capable devices on hand.

chrisdutz, 2024-09-23:
Hehe ... ok ... I guess that's really the only case it would make sense ;-)

As I also have a Beckhoff ADS device here ... if you could describe and document how to setup the EIP stuff, I could give it a try on my device.

JoonasC, 2024-09-23:
Sure, here is Beckhoff's official documentation: [https://infosys.beckhoff.com/english.php?content=../content/1033/tf6281_tc3_ethernetipscanner/index.html&id=](https://infosys.beckhoff.com/english.php?content=../content/1033/tf6281_tc3_ethernetipscanner/index.html&id=)

For a TLDR version:
1. Add an Ethernet/IP scanner device:
[image]
2. Create a sync task for the device:
[image]
3. Create a global variable list called `ETHIP`, which contains variables declared in the `M` memory area:
[image]
4. Add the `Tc2_EthernetIP` library to the PLC project:
[image]
5. Check the read and write options for tags you want to read/write to:
[image]
6. Get the IP address of the Ethernet/IP server from the settings tab:
[image]