apache/plc4x #1802: [Bug]: Mitsubishi MXOPC UA Server not connecting with Plc4j.

codekundan, 2024-10-06:
### What happened?

Hi Plc4j Team,

I am using **Mitsubishi MXOPC UA Server**. I tried to connect it with Plc4j driver but its throwing error. I even tried making _"Security Policy : None" and "Message Security Policy : None"_, but still it was not connecting.

Below is the code 

`package com.mycompany.plc4xconnect;

import org.apache.plc4x.java.api.PlcConnection;
import org.apache.plc4x.java.api.PlcDriverManager;
import org.apache.plc4x.java.api.exceptions.PlcConnectionException;

public class Plc4xConnect {

    public static void main(String[] args) throws PlcConnectionException, Exception {
        String connectionString = "opcua:tcp://localhost:4841";
        
        try (PlcConnection plcConnection = PlcDriverManager.getDefault().getConnectionManager().getConnection(connectionString)) {
           System.out.println("Connection established..."+plcConnection);
           if(!plcConnection.getMetadata().isWriteSupported()) {
               System.out.println("This connection doesn't support writing");
           } else {
               System.out.println("This connection supports writing");
           }
        }
    }
}`

Below is the Code Output

[image]

Below is the pic of MXOPC UA Server 

[image]

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

mrwhy-orig, 2024-10-06:
Could you please set your logging framework to Debug and add the output here? Thanks!

splatch, 2024-10-07:
Is it possible to obtain demo installation of MXOPC server?

codekundan, 2024-10-07:
Below is the Demo installation video of Mitsubishi MXOPC UA server. 

Best Regards
[]

https://github.com/user-attachments/assets/c39f0267-b303-4e10-b62e-90ce29defc46

codekundan, 2024-10-07:
[image]

The Connection String (marked in red circle), I tried to connect with it i.e. **"opc.tcp://localhost:4841"**, but it gives error : **"Unable to find driver for protocol 'opc.tcp' "**. So i used the Connection String as **"opcua:tcp://localhost:4841"**. It gave **"Connection terminated by remote"** error.

Regards
Kundan Negi

codekundan, 2024-10-07:
By Default, it uses **"Security Policy : Basic128Rsa15"** and **"Message Security Mode : Sign & Encrypt"**.

codekundan, 2024-10-07:
@splatch I have sent the MXOPC UA Server in your luke@code-house.org mail. Please check it.

codekundan, 2024-10-15:
@splatch Did Plc4j connect with MXOPC UA Server?

splatch, 2024-10-16:
@codekundan I've managed to install it, but haven't had time to configure it and test connectivity itself.

splatch, 2024-10-16:
After disabling security I've managed to connect. I could not confirm yet, if encrypted communications works as expected.

My connection string is `opcua:tcp://DESKTOP-OM18VQ9:4841?discovery=true&security-policy=NONE&message-security=NONE`.
I've used simulated OPC tag called `Address Space.Dev00.Tag000`.

Test code:
```
        try (PlcConnection connection = new DefaultPlcDriverManager().getConnection("opcua:tcp://DESKTOP-OM18VQ9:4841?discovery=true&security-policy=NONE&message-security=NONE")) {
            CompletableFuture<? extends PlcSubscriptionResponse> future = connection.subscriptionRequestBuilder()
                .addChangeOfStateTagAddress("c", "ns=4;s=Address Space.Dev00.Tag000")
                .build().execute();
            future.whenComplete((r, e) -> {
                if (e != null) {
                    e.printStackTrace();
                    return;
                }
                r.getSubscriptionHandle("c").register(event -> {
                    for (String field : event.getTagNames()) {
                        System.out.println("Received state update: " + field + " " + event.getObject(field) + " " + event.getResponseCode(field));
                    }
                });
            }).get();
            Thread.sleep(600_000);
            System.out.println("Closing connection");
        } catch (PlcConnectionException e) {
            throw new PlcRuntimeException(e);
        } catch (Exception e) {
            e.printStackTrace();
        }
```

splatch, 2024-10-17:
I followed the secure connection issues, it turns that our encryption logic doesn't work if client and server use keys with different length. By default MX OPC offers 1024 bit key, whereas client logic by default generates 2048 bit key.

splatch, 2024-10-17:
@codekundan I've pushed a fix so MX OPC in its default configuration can work, please give a try to develop version (0.13-SNAPSHOT). I just confirmed that our client works with 1024 bit certificate used by this server.

codekundan, 2024-10-18:
@splatch I developed 0.13.0-SNAPSHOT but I am unable to connect. I did connect when Unsecured Connections are turned on. But for Unsecured Connections off, i am getting same **Connection terminated at Remote** error.  Can you please share the Connection String and the console output?

codekundan, 2024-10-19:
@splatch How to generate Plc4x certificate? I did try to copy der file in CA/rejected to CA/certs directory but still cannot connect. And you are correct that every time a new ad-hoc certificate is generated at CA/rejected directory. Also i cannot make out whether the certificate generated in CA/rejected is Plc4x certificate or ad-hoc certificate. Sorry for bothering a lot.

codekundan, 2024-10-19:
@splatch I downloaded Unified-Automation UaExpert Client. It too created certificated in CA/rejected directory. I cut and pasted it to CA/certs directory. And it seemed to work. Is it the same way I have to create Plc4x certificate by executing the connection code which then creates a certificate in CA/rejected directory. And then cut and paste it to CA/certs directory. Am I doing it correct?

splatch, 2024-10-21:
Hello @codekundan you probably can use same certificate for plc4j, you just need to encode it correctly. Eventually use https://keystore-explorer.org/ to create java keystore which you can wire through client configuration parameters areL keystore-file, keystore-password (if provided) and keystore-type (if different than pkcs12).

codekundan, 2024-10-28:
Hello @splatch I tried creating a certificate with KeyStore Explorer but failed to connect. I created a Certificate with "**Public key : RSA 1024 bits", "Signature Algorithm : SHA-1 with RSA", "Fingerprint : SHA-1"**. Then I created a **PKCS#1 Private Key with OpenSSL**. I kept Certificate in CA/certs directory and Private Key in CA/private directory.  But could not connect. Also I noticed when doing all this if I open MX OPC UA Configurator Client then it shows error message. Then if I delete my custom created Certificate and private key from respective directories, MXOPC UA Configurator Client Tool gets connected to MX OPC UA Server again. Also I tried wiring my custom created KeyStore in the Connection String using parameters like key-store-file, key-store-type=pkcs12 but failed to connect. Can you please share how you created your custom certificate and private key with 1024 bits and connected.

codekundan, 2024-10-29:
Thanks @splatch . Will let you know.

splatch, 2024-10-29:
Website update is deployed, so you can use rendered version: https://plc4x.apache.org/users/getting-started/opcua-client-certificate.html :-)

codekundan, 2024-10-30:
Hello @splatch I tested it. It is working in my PC. I just need one more week to actually test it in Mitsubishi PLC as currently Diwali Festival Holidays are going on in India. After that we can close this issue as solved.

splatch, 2024-11-06:
Closing as issue was confirmed to be solved.

closed 2024-11-06 by splatch

codekundan, 2024-11-06:
Hello @splatch Thanks a lot.