apache/plc4x #1856: [Bug]: PLC4PY, Write a variable integer value with python and modbus

telmobarrena98, 2024-10-29:
### What happened?

I am trying to write an integer variable value with python and modbus with the following code:
```
with connection.write_request_builder() as builder:
            builder.add_item("Random Tag 1", "4x00001", 2)
            request = builder.build()
```
However, I am receiving this error: 
`AttributeError: 'int' object has no attribute 'get_int'`
Changing the DataItem.py 583 line, from:
`value: int = _value.get_int()`
to:
`value: int = _value`
makes the code work, is it a bug? or should I convert the integer value in 
`builder.add_item("Random Tag 1", "4x00001", 2)`
to any type?

### Version

v0.13

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

ottlukas, 2024-10-29:
Try wrapping the value in a PlcINT, so PlcINT(2)

hutcheb, 2024-10-29:
Not sure what how my comment ended up being posted by @ottlukas, might need to talk to GH about how that happened.

@telmobarrena98 I'll update the documentation when I get a chance to reflect the change in how values are passed to write requests.

Let us know how you get along with it.

ottlukas, 2024-10-30:
> Not sure what how my comment ended up being posted by @ottlukas, might need to talk to GH about how that happened.
> 
> @telmobarrena98 I'll update the documentation when I get a chance to reflect the change in how values are passed to write requests.
> 
> Let us know how you get along with it.

Apologies this was due that your response via mailing list (Subscribe to [dev@plc4x.apache.org](mailto:dev@plc4x.apache.org).)  I reposted without stating that I reposted to GitHub as the response was not posted here in GitHub. Should have mentioned that in the beginning.

telmobarrena98, 2024-10-30:
Solved thanks!

closed 2024-10-30 by telmobarrena98