apache/plc4x #1859: [Bug]: GO MODBUS invalid read request

telmobarrena98, 2024-10-30:
### What happened?

After successful connection via MODBUS with GO, I try to get a variable value: 
```
// Prepare a read-request
readRequest, err := connection.ReadRequestBuilder().
	AddTagAddress("tag1", "holding-register:1:REAL").
	Build()
fmt.Printf("err.Error(): %v\n", err.Error())
```
, however I am receiving this output:
```
{"level":"debug","time":"2024-10-30T13:55:25+01:00","message":"No quantity supplied, assuming 1"}
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x18 pc=0x4f7397]
```

### Version

V1.23.2

### Programming Languages

- [ ] plc4j
- [X] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

sruehl, 2024-10-30:
usually there should be a stack trace on a panic attached. Check if you can attach that too

telmobarrena98, 2024-10-30:
I was missing to put the "err" inside an if statement, like this:
`
if err != nil {
fmt.Printf("err.Error(): %v\n", err.Error())
}
`
, solved thanks

closed 2024-10-30 by telmobarrena98