apache/plc4x #1864: [Bug]: Error when connecting to OPCUA server with plc4go

telmobarrena98, 2024-11-04:
### What happened?

I am launching an OPCUA server through Docker running the following .py file:
```
import asyncio
import logging
import random
from asyncua import Server, ua

async def main():
    _logger = logging.getLogger("asyncua")
    # setup our server
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = await server.register_namespace(uri)
    ns = "ns=2;s=freeopcua.Tags.pressure"
    ns2 = "ns=2;s=freeopcua.Tags.temperature"
    
    min_val = -0.5
    max_val = 0.6

    # populating our address space
    # server.nodes, contains links to very common nodes like objects and root
    myobj = await server.nodes.objects.add_object(idx, "MyObject")
    pressure = await myobj.add_variable(ns, "MyVariable", 10.5)
    temperature = await myobj.add_variable(ns2, "MyVariable", 26.7)
    # Set MyVariable to be writable by clients
    await pressure.set_writable()
    await temperature.set_writable()
    opcs = [ pressure, temperature ]
  
    _logger.info("Starting server!")
    async with server:
        while True:
            await asyncio.sleep(100)
            random_counter = random.uniform(min_val, max_val)
            for opc in opcs:
                new_val = await opc.get_value() + random_counter
                if(new_val>100.0):
                    new_val=100.0
                elif(new_val<0.0):
                    new_val=0.0
                _logger.info("Set value of %s to %.1f", opc, new_val)
                await opc.write_value(new_val)

if __name__ == "__main__":
    logging.basicConfig(level=logging.DEBUG)
    asyncio.run(main(), debug=True)
```

. I am able to connect to it with UA Expert. However, when using the opcua driver of plc4go, i get this error in the server side: 
`    raise ua.UaError(f"Unsupported message type {header.MessageType}")
asyncua.ua.uaerrors._base.UaError: Unsupported message type b'\x00\x00\x00`
and this one on the .go file:
`
Error connecting to PLC: Error during connection: timeout after 5s
`

### Version

v1.23.1

### Programming Languages

- [ ] plc4j
- [X] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

splatch, 2024-11-05:
Golang driver is not maintained and requires major effort to align it with java version. The encryption part for go is not implemented beyond initial handshake.

telmobarrena98, 2024-11-05:
Okey thanks @splatch , is plc4py working for OPC-UA? In the documentation I see that is not implemented.

splatch, 2024-11-05:
I believe python currently covers modbus and umas, AFAIK it does not support python, but you can confirm that with @hutcheb and @ottlukas.

hutcheb, 2024-11-05:
Yes, you are correct it isn’t implemented in plc4py.

closed 2024-11-07 by telmobarrena98