apache/plc4x #1881: [Feature Request]: Support OPC UA options

tvwy, 2024-11-09:
### What would you like to happen?

I wanted to try out the OPC UA driver, but I'm unsure if I'm making a major mistake or if parsing string options simply hasn't been implemented yet. Following the documentation it seems to be implemented at least: https://plc4x.apache.org/users/protocols/opcua.html

```
package main

import (
	"fmt"

	plc4go "github.com/apache/plc4x/plc4go/pkg/api"
	"github.com/apache/plc4x/plc4go/pkg/api/drivers"
	"github.com/apache/plc4x/plc4go/pkg/api/transports"
)

func main() {
	// Create a new instance of the PlcDriverManager
	driverManager := plc4go.NewPlcDriverManager()

	// Register the Transports
	transports.RegisterTcpTransport(driverManager)
	transports.RegisterUdpTransport(driverManager)

	// Register the Drivers
	drivers.RegisterOpcuaDriver(driverManager)

	// Set connection options, including username and password
	connectionRequestChannel := driverManager.GetConnection("opcua:tcp://127.0.0.1:7894/UA/Plant?username=user1&password=password1")
	// Wait for the connection to be established
	connectionResult := <-connectionRequestChannel

	// Check if something went wrong
    if connectionResult.GetErr() != nil {
	    fmt.Printf("Error connecting to PLC: %s", connectionResult.GetErr().Error())
		return
	}

	connection := connectionResult.GetConnection()

	// Make sure the connection is closed at the end
	defer connection.Close()
```

resulting in: 

```
 go run main
{"level":"trace","time":"2024-11-09T14:58:33+01:00","message":"Creating plc driver manager"}
{"level":"debug","transportName":"TCP/IP Socket Transport","time":"2024-11-09T14:58:34+01:00","message":"Registering transport"}
{"level":"info","transportName":"TCP/IP Socket Transport","time":"2024-11-09T14:58:34+01:00","message":"Transport for transportName registered"}
{"level":"debug","transportName":"UDP Datagram Transport","time":"2024-11-09T14:58:34+01:00","message":"Registering transport"}
{"level":"info","transportName":"UDP Datagram Transport","time":"2024-11-09T14:58:34+01:00","message":"Transport for transportName registered"}
{"level":"debug","protocolName":"Opcua","time":"2024-11-09T14:58:34+01:00","message":"Registering driver"}
{"level":"info","protocolName":"Opcua","time":"2024-11-09T14:58:34+01:00","message":"Driver for protocolName registered"}
{"level":"debug","transportName":"TCP/IP Socket Transport","time":"2024-11-09T14:58:34+01:00","message":"Registering transport"}
{"level":"warn","transportName":"TCP/IP Socket Transport","time":"2024-11-09T14:58:34+01:00","message":"Transport already registered"}
{"level":"debug","connectionString":"opcua:tcp://127.0.0.1:7894/UA/Plant?username=user1&password=password1","time":"2024-11-09T14:58:34+01:00","message":"Getting connection for connectionString"}
{"level":"debug","connectionUrl":"opcua:tcp://127.0.0.1:7894/UA/Plant?username=user1&password=password1","time":"2024-11-09T14:58:34+01:00","message":"parsed connection URL"}
{"level":"debug","connectionUrl":"opcua:tcp://127.0.0.1:7894/UA/Plant?username=user1&password=password1","protocolName":"Opcua","time":"2024-11-09T14:58:34+01:00","message":"got driver protocolName"}
{"level":"trace","time":"2024-11-09T14:58:34+01:00","message":"we handling a opaque connectionUrl"}
{"level":"debug","transportName":"tcp","transportConnectionString":"127.0.0.1:7894","time":"2024-11-09T14:58:34+01:00","message":"got a transport"}
{"level":"debug","transportUrl":"tcp://127.0.0.1:7894/UA/Plant","time":"2024-11-09T14:58:34+01:00","message":"Assembled transport url"}
{"level":"debug","transportUrl":"tcp://127.0.0.1:7894/UA/Plant","numberTransports":2,"numberDriverOptions":2,"time":"2024-11-09T14:58:34+01:00","message":"Get connection for transport url"}
**Error connecting to PLC: can't parse options: string not yet supported**
```

Source of behaviour is the ParseFromOptions function in Configuration.go of opcua not having implemented a string case.
[image]

It's a Feature Request because I'm not sure.

### Programming Languages

- [ ] plc4j
- [X] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

splatch, 2024-11-11:
All I can say is - golang driver for opcua is far behind java version.

tvwy, 2024-11-11:
Alright. Thanks for the quick reply!

closed 2024-11-11 by tvwy