apache/plc4x #1894: [Feature Request]: Modbus TCP connection pooling

IsmoLeszczynski, 2024-11-15:
### What would you like to happen?

When using Modbus TCP gateways, the gateways can aggregate multiple Modbus devices using unit ids. Such gateways unfortunately also seem to sometimes have limits on how many TCP connections can be active at the same time, due to which forming a connection per unit id is not feasible. Is there a way in PLC4X to share the TCP connection between different connection strings? I know of the connection cache, but it caches per connection string, which is different per unit id in this case.

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-11-15:
You should have a look at our connection cache: https://plc4x.apache.org/plc4x/0.13.0-SNAPSHOT/users/tools/connection-cache.html This should be a general purpose solution for your problem for all PLC4X drivers.

IsmoLeszczynski, 2024-11-15:
Thanks for the reply, I was wondering how to make use of that with different unit ids, but then I noticed I somehow skipped a line in the Modbus documentation where the unit id can be given per Modbus address. Apologies for the hassle!

closed 2024-11-15 by IsmoLeszczynski

chrisdutz, 2024-11-15:
No need to appologise :-)

splatch, 2025-04-14:
> Thanks for the reply, I was wondering how to make use of that with different unit ids, but then I noticed I somehow skipped a line in the Modbus documentation where the unit id can be given per Modbus address. Apologies for the hassle!

@IsmoLeszczynski You can make unit-id part of tag, since it is not needed for negotiation and is used for each and every packet we submit:
https://plc4x.apache.org/plc4x/pre-release/users/protocols/modbus.html#_general_format

The unit-id set at the connection level serves as default, tag option overrides it. This is new feature we still have only in develop.