apache/plc4x #1895: [Bug]: DataType WORD, DWORD and WSTRING not supported in OPC-UA.

codekundan, 2024-11-15:
### What happened?

Hi Plc4x Team and @splatch . I was testing Plc4x with Mitsubishi PLC. I was not able to read and write **WORD, DWORD and WSTRING datatype**. It gave me **"Datatype WORD is unsupported by this protocol" error** as can be seen in pic below:-

[image]

I want to ask whether OPC-UA especially Plc4j supports these 3 Datatypes (WORD, DWORD, WSTRING)? I was able to read and write with other Datatypes such as INT, REAL, BOOL, STRING. Any help regrading this ?

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

splatch, 2024-11-16:
Hello @codekundan, I think this type is not recognized at the protocol level, but defined on client side. Which type from specification are you actually trying to read: https://reference.opcfoundation.org/Core/Part6/v104/docs/5.1.2?

codekundan, 2024-11-18:
Hello @splatch , I am trying to test the available Datatypes in the PLC4X OPC-UA website https://plc4x.apache.org/plc4x/0.13.0-SNAPSHOT/users/protocols/opcua.html#:~:text=UINT%20(uint%2016)-,WORD%20(uint%2016),-DINT%20(int%2032

If you look at the link, Datatypes like WORD, DWORD and WSTRING are mentioned. So I thought they should be supported.

codekundan, 2024-11-19:
Hello @splatch , I am closing this issue. If you want to add something please feel free to open it.

closed 2024-11-19 by codekundan