apache/plc4x #1925: [Feature Request]: Failed to read a V area data from a S7-200 Smart

typhoon322, 2024-12-04:
### What would you like to happen?

Failed to read a V area data from a Simens S7-200 Smart due to the MemoryArea Enum doesn't define one, the address is like VW310 or V350.0. 
[image]

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

tingbotan, 2024-12-20:
same question

typhoon322, 2024-12-22:
> same question

replace “V” with "DB1.DB" , it worked well.

closed 2024-12-22 by typhoon322

chrisdutz, 2024-12-22:
Good catch:
https://www.fernhillsoftware.com/help/drivers/siemens-s7/data-address-format.html
"V Memory | Vnnnn | V is an alias for Data Block 1."
Possibly we could make the driver understand this alias ... same as the "E" and "A" (german) alternatives for Q and I

typhoon322, 2024-12-22:
> Good catch: https://www.fernhillsoftware.com/help/drivers/siemens-s7/data-address-format.html "V Memory | Vnnnn | V is an alias for Data Block 1." Possibly we could make the driver understand this alias ... same as the "E" and "A" (german) alternatives for Q and I

thanks for sharing this, bro!

john1337, 2025-09-06:
> ### What would you like to happen?
> Failed to read a V area data from a Simens S7-200 Smart due to the MemoryArea Enum doesn't define one, the address is like VW310 or V350.0. [image]
>
> ### [...]

in my case ,connection can't be made when using s7-200,how did you do it ,my example code 
            PlcConnection connection = PlcDriverManager.getDefault()
                    .getConnectionManager()
                    .getConnection("s7://192.16.8.249?controller-type=S7_200&transport-protocol=pniop&remote-rack"
                            + "=0&remote-slot=2");
and i am using these two following maven dependencies:
    <dependency>
      <groupId>org.apache.plc4x</groupId>
      <artifactId>plc4j-api</artifactId>
      <version>0.13.0</version>
    </dependency>
    <dependency>
         <groupId>org.apache.plc4x</groupId>
         <artifactId>plc4j-driver-s7</artifactId>
         <version>0.13.0</version>
     </dependency>