apache/plc4x #1940: [Feature Request]: EtherNet/IP Implicit Messaging is supported ?

nqduy35, 2024-12-17:
### What would you like to happen?

I wonder if the EtherNet/IP Implicit Messaging is supported in the current version (0.12.0) of plc4j-driver-eip ? 

If not, is it a good idea to use the plc4j API just to create a EtherNet/IP connection, then writing some  code to read message at port 2222 and handle the CPF data by myself ?

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-12-23:
So far implicit messaging doesn't seem to have been implemented yet. 
As far as I can see it, the implicit messaging would correlate nicely to our Subscription API. Also would our connection concept support an UDP "connection". So in theory the ideal solution would be that you implement the missing subscription functionality as part of the PLC4X EIP driver and work with the rest of the PLC4X community to finish it?

That would be more sustainable from your perspective as otherwise you'll have to maintain your functionality manually, by doing it the open-source way, you sort of get maintenance for free.

chrisdutz, 2024-12-23:
Some possibly interesting info for anyone willing to work on this:
https://library.automationdirect.com/ethernetip-implicit-vs-explicit-messaging/