apache/plc4x #1950: Make the last released version the default of our website

chrisdutz, 2024-12-22:
### What would you like to happen?

Curently the default of the website guides people to use the SNAPSHOT version of PLC4X. We should change this to show the last released version per default and give people the chance to update to the SNAPSHOT version of it.

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

hutcheb, 2024-12-22:
So that would mean that we would be using the v0.12 branch as the latest version on the website, and then having a seperate pre-release version (I want to avoid using the snapshot names in the urls as it is only temporary).

It also means any changes to the develop branch website will show up on the pre-release version.

That should just be a matter of changing the name of the branches in the config file.

chrisdutz, 2024-12-23:
Yeah .... I think that would be best ... usually it's common at apache to refer to the documentation on the latest release, not the latest development version.

hutcheb, 2024-12-28:
That has been updated

closed 2024-12-28 by hutcheb