apache/plc4x #1972: [Bug]: PlcSubscriptRequest class does not have this method addEventField

rooterWzs, 2025-01-10:
### What happened?

[image]

### Version

0.13.0-SNAPSHOT

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

rooterWzs, 2025-01-10:
[image]

chrisdutz, 2025-01-10:
This is a documentation bug and not a bug in the API. We refactored the API for subscriptions quite a bit quite some time ago. Look here: https://github.com/apache/plc4x-extras/blob/develop/plc4j/examples/plc4j-s7event/src/main/java/org/apache/plc4x/examples/plc4j/s7event/PlcCycSubscriptionS7400H.java#L287

@ceos01 could you please update the code on your S7 documentation page?

closed 2025-01-13 by rooterWzs