apache/plc4x #1974: [pick-brains][code-generation] add omron fins protocol

rooterWzs, 2025-01-14:
I created the Omron protocol in plc4x/protocols and created the Omron module in plc4j/drivers. When I ran mvn install - P update generated code with java, an error occurred, as shown in the figure，May I ask what caused this and how can I resolve it？

[image]

[image]

[image]

[ERROR] Failed to execute goal org.apache.plc4x.plugins:plc4x-maven-plugin:1.8.0-SNAPSHOT:generate-driver (generate-driver) on project plc4j-driver-omron: Unable to find protocol specification module 'omron' with version 'undefined' on modules classpath -> [Help 1]

hutcheb, 2025-01-14:
It looks like you might have some config incorrect in your pom. Looks like you might be trying to build a module with 0.13.0 as the version instead of 0.13.0-SNAPSHOT.

But if you could push your changes somewhere we can see if anything stands out.

chrisdutz, 2025-01-14:
Do you have your repo available anywhere? I could have a look.

rooterWzs, 2025-01-14:
> It looks like you might have some config incorrect in your pom. Looks like you might be trying to build a module with 0.13.0 as the version instead of 0.13.0-SNAPSHOT.
> 
> But if you could push your changes somewhere we can see if anything stands out.

> Do you have your repo available anywhere? I could have a look.

https://github.com/rooterWzs/plc4x/tree/develop

chrisdutz, 2025-01-14:
Ok ... I had a look and there were several issues:
1. You name the protocol "fins" in protocols/omron/src/main/java/org/apache/plc4x/protocol/omron/FinsProtocol.java but you try to use it as "omron" in plc4j/drivers/omron/pom.xml ... the two have to match, I tried setting it to "fins" and it picked up the protocol definition
2. Your mspec had multiple issues:
    - There was an odd prefix to the mspec: "好的，根据你提供的西门子PLC S7协议的MSpec文件结构，我将为你生成一个详细的Omron FINS协议的MSpec文件。这个文件将包括FINS协议的基本消息结构、请求和响应消息、以及一些常见的命令和响应。

### Omron FINS协议MSpec文件

```freemarker" ... that needed to be removed. 
     - You repeatedly defined byte-arrays like this:
             [simple byte[] data length 'wordCount * 2'] 
        However this needed to be changed to: 
            [array  byte    data length 'wordCount * 2']
        (Notice the keyword "array" instead of "simple"? Also the bracets needed to be removed
     - In line 192 (187 after removing the header) you define a string simple type ... strings need a fixed number of bits ... vstring doesn't, but that requires an expression on how long the string should be
     - Your enum FINSResponseCode had one additional closing square bracket, that needed to be removed

With these changes I was able to generate the code for your mspec and compile it.

I did add a config option to skip the maven-dependency-plugin execution as you're just starting to implement your driver ... no need to get too confused about that. You can take care of that once you're done.

chrisdutz, 2025-01-14:
[fix__Got_the_fins_code-generation_working_.patch](https://github.com/user-attachments/files/18412196/fix__Got_the_fins_code-generation_working_.patch)

Here's a patch file with all of my changes.
Hope that helps.

rooterWzs, 2025-01-15:
MVN can pass, but I still need to understand the generation rules of MSPEC. Do you have any good suggestions

chrisdutz, 2025-01-15:
Well I guess having a look at the other protocol mspecs as inspiration would help. Also possibly the mspec in the code-Gen testsuite (test mspec)

closed 2025-01-16 by rooterWzs