apache/plc4x #1979: [pick-brains][omron-fins.mspec] Regarding the message body

rooterWzs, 2025-01-19:
I have a question to ask you: Should I encapsulate it into one message body or multiple message bodies?
**example one body:**
// FINS Message
[discriminatedType FINSMessage
    [const              uint 32             protocolId      0x46494E53]             // FINS协议标识符
    [implicit           uint 32             length          'lengthInBytes']               // 数据包长度，自动计算
    [discriminator      uint 32             command]
    [const              unit 32             errorCode       0x00000000]
    [typeSwitch command
        ['0x00000000' FINSHandshakeRequest
        ]
        ['0x00000001' FINSReadRequest
        ]
        ['0x00000002' FINSWriteRequest
        ]
    ]
    [optional FINSParameter ('command')]
]

**example multiple message bodies:**

// FINSRequestMessage
[discriminatedType FINSRequestMessage
.........
]

// FINSResponseMessage
[discriminatedType FINSResponseMessage
.........
]

chrisdutz, 2025-01-19:
One ask, that I have for you: Could you please update your comments to english?

But to your question: Your first option looks absolutely standard. It's also exactly the use case we built the discriminated types for. I would definitely reccommend doing that.

Also we recently added the ability of nested typeSwitches. In the past you couldn't nest these and you needed to add additional intermediate types. Now you could even have for examples multiple types of FINSReadRequests which share some common properties.

If the protocol doesn't provide the information needed to decide between request and response, you can also pass in arguments from the driver (Have a look in the Modbus Protocol).

PS: I think it would most probably be best to not open issues for this type of discussion, but to open a discussion in the discussion section.

rooterWzs, 2025-01-19:
Firstly, thank you very much for giving me direction. I will continue to improve.
Okay, I will translate the annotations into English.
I once tried to send an email to dev@plc4x.apache.org But I haven't received a reply, why is that?

rooterWzs, 2025-01-19:
Firstly, thank you very much for giving me direction. I will continue to improve.
Okay, I will translate the annotations into English.
I once tried to send an email to [dev@plc4x.apache.org](mailto:dev@plc4x.apache.org) But I haven't received a reply, why is that?

closed 2025-01-19 by rooterWzs

chrisdutz, 2025-01-19:
Hmmm odd... Don't recall that email and didn't get any email for moderation.