apache/plc4x #1997: [Bug]: Modbus Over TCP Reading Float Value in Negative Issue

Balaji-Sundar, 2025-02-01:
### What happened?

Modbus Over TCP Reading Float Value in Negative Issue. Actual value is 1.93e+07
public static void main(String[] args) {
try {
String ip = "10.7.10.1";
int port = 502;
int slaveId = 2;
int requestTimeOut = 10000;
String payLoadByteOrder = "LITTLE_ENDIAN_BYTE_SWAP";
String connectionUrl = String.format("modbus-tcp:tcp://%s:%d?unit-identifier=%d&request-timeout=%d&default-payload-byte-order=%s", ip, port, slaveId, requestTimeOut, payLoadByteOrder);
try (PlcConnection plcConnection = new DefaultPlcDriverManager().getConnection(connectionUrl)) {
if (!plcConnection.getMetadata().isReadSupported()) {
System.out.println("This connection doesn't support reading.");
}
PlcReadRequest.Builder builder = plcConnection.readRequestBuilder();
builder.addTagAddress("value1", "holding-register:159:REAL[2]");
PlcReadRequest readRequest = builder.build();
PlcReadResponse response = readRequest.execute().get();
System.out.println("response = " + response.toString());
for (String tagName : response.getTagNames()) {
System.out.println("Tag: " + tagName);
System.out.println("Response code for " + tagName + ": " + response.getResponseCode(tagName));
int numValues = response.getNumberOfValues(tagName);
System.out.println("numValues = " + numValues);
if (response.getResponseCode(tagName) == PlcResponseCode.OK) {
System.out.println("response = " + response.getTag(tagName));
System.out.println("response = " + response.getFloat(tagName));
//response.getAllBytes(tagName).stream().map(HexUtil::toHex).map(hex -> "Register Value: " + hex).forEach(System.out::println);
} else {
System.out.println("Error[" + tagName + "]: " + response.getResponseCode(tagName).name());
}
}
}
} catch (Exception e) {
System.out.println("e = " + e);
}
}
Output:
response = org.apache.plc4x.java.spi.messages.DefaultPlcReadResponse@6a47b187
Tag: value1
Response code for value1: OK
numValues = 2
response = ModbusTagHoldingRegister {address=158, quantity=2, dataType=REAL }
response = -7.128774E35

### Version

v0.12.0

### Programming Languages

- [x] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [x] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2025-02-01:
Well. Modbus technically doesn't know floats. We just interpret four bytes as a float. Possibly your system uses a different encoding?

What type of device are you using?

Balaji-Sundar, 2025-02-01:
We are using USR-TCP232-410S device to query RS-485 supported modbus device.

    Baud Rate : 9600
    Data Size : 8bit
    Parity : Even
    Stop bits : 1
    Flow Control : None

Regards

-S.Balaji

chrisdutz, 2025-02-01:
Could you please do a wireshark capture of your attempt to read the float value as well as which value you expect it to be? Then I think we could have a look.

Balaji-Sundar, 2025-02-03:
I find out the some of energy meters having "LITTLE_ENDIAN_BYTE_SWAP" encoding. Current version does not have encoding support. When can we have 0.13.0 release.

chrisdutz, 2025-02-03:
Could you please double check, if it works with the latest development version? There were plans to do a new release pretty soon.

Balaji-Sundar, 2025-02-04:
I have tried 0.13.0.SNAPSHOT version. I am getting Timeout exception.
String ip = "10.7.10.1";
            int port = 502;
            int slaveId = 2;
            int requestTimeOut = 5000;
            String payLoadByteOrder = "LITTLE_ENDIAN_BYTE_SWAP";
            //String connectionUrl = String.format("modbus-tcp://%s:%d?unit-identifier=%d&request-timeout=%d&default-payload-byte-order=%s", ip, port, slaveId, requestTimeOut, payLoadByteOrder);
.addTagAddress("value1", "holding-register:159:REAL[2]{byte-order:'LITTLE_ENDIAN_BYTE_SWAP'}")

chrisdutz, 2025-02-04:
I guess without a wireshark capture, I'd just be guessing.