apache/plc4x #2005: [Bug]: S7 connection isConnected doesn't update on disconnect

IsmoLeszczynski, 2025-02-10:
### What happened?

I've been running the 0.13.0-SNAPSHOT version against my S7-1500 and I've noticed that, most of the time, the `S7HPlcConnection.isConnected()` doesn't update to reflect a clear disconnect, e.g. dropping of the network connection completely between my laptop and the PLC, if an initial connection has succeeded before the disconnect. The only time it seems to update is when a netty exception occurs during a reconnect, which I haven't managed to replicate reliably.

My connection string, IP omitted: `s7://x.x.x.x?controller-type=S7_1500`

While investigating the case, I also tried to do a manual `S7HPlcConnection.ping()` to see what it's result is, but noticed that the method always returns `null` for the `S7HPlcConnection` - not sure what the reason is, but it seems odd that the method is defined as `CompletableFuture<? extends PlcPingResponse>`, but the S7 implementation will never return the real result, only log it. Might be some history that I'm not aware of, but I was directed to it from the Connection Cache document that mentioned `ping()` being used to check the validity of connections.

Is there a known reason why the `isConnected()` doesn't update after an initial connection succeeds, or am I missing something here?

### Version

0.13.0-SNAPSHOT

### Programming Languages

- [x] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [x] S7

glcj, 2025-02-10:
Hi,

I'm dealing with an emergency for a client right now, but I'm interested in solving this S7 driver issue.

Please try this commit

6037919a3e6a1778f47f04fafac2a532a51879fc

After this commit, changes were made that affected the driver.

If it doesn't work for you, I'll keep an eye on the details.

Kind regards,

IsmoLeszczynski, 2025-02-10:
I've been testing this a bit more and the state actually changed, but it took longer than I expected (~60s) with default parameters, and I probably didn't wait that long when flicking the switch on and off. I had also tried to use the read-timeout parameter alone as its relationship to the ping settings wasn't very clear from the docs, I found the relationship from the sources when digging for the timeout. As soon as I set both the read-timeout and the ping parameters, I got the shorter timeout that I had expected as well.

So in the end, the state does change, but the connection string parameters affecting the timeout length are a bit unclear in the docs.

closed 2025-02-10 by IsmoLeszczynski