apache/plc4x #2013: [Bug]: Issue with connection using 4096-bit certificates

gustavoschewinski, 2025-02-17:
### What happened?

Hello,

The connection to PLC4X using 2048-bit certificates (both client and server) works fine. However, when switching to 4096-bit certificates for both, the connection fails.

On the OPCUA side, I receive an error stating "An error occurred verifying security", and in the PLC4X logs, I only see "Connection terminated by remote." It appears that the certificates are not being accepted.

I tested the certificates separately by doing a connection using a python library, and it worked, so the issue does not seem to be with the certificates themselves.

Any insights or suggestions would be greatly appreciated on how to fix this problem.

Thanks a lot!

### Version

v0.13.0-SNAPSHOT

### Programming Languages

- [x] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [x] OPC-UA
- [ ] S7

splatch, 2025-03-03:
Hello @gustavoschewinski can you provide commands for creation of certs? I would then have a reproducer locally to see why there is failure. We have some unit tests which confirm handling of specific key lengths, so if they do not spot issue its most likely other part of negotiation logic which is out of sync with spec.

gustavoschewinski, 2025-03-03:
Hello @splatch, 

Thanks for helping, this is how I am creating them:

1. **CA Certificate Creation**  
   Create a configuration file `ca.conf` and use it to generate a CA certificate.

   `ca.conf`:

   ```plaintext
   [ req ]
   default_bits = 4096
   default_md = sha256
   distinguished_name = Digilab
   x509_extensions = v3_ca
   prompt = no

   [ Digilab ]
   countryName = DE
   stateOrProvinceName = Bayern
   localityName = Munich
   organizationName = MTU
   commonName = FIXI

   [ v3_ca ]
   basicConstraints = CA:TRUE, pathlen:0
   keyUsage = keyCertSign, cRLSign
   subjectKeyIdentifier = hash
   authorityKeyIdentifier = keyid:always,issuer:always

   [ ca ]
   default_ca = CA_default

   [ CA_default ]
   dir             = ./myCA
   database        = $dir/index.txt
   new_certs_dir   = $dir/certs
   certificate     = $dir/ca_cert.pem
   serial          = $dir/serial
   crlnumber       = $dir/crlnumber
   crl             = $dir/ca_crl.pem
   private_key     = $dir/private/ca_key.pem
   RANDFILE        = $dir/private/.rand

   default_md      = sha256
   policy          = policy_anything

   default_crl_days = 300

   [ policy_anything ]
   countryName             = optional
   stateOrProvinceName     = optional
   localityName            = optional
   organizationName        = optional
   commonName              = supplied
   emailAddress            = optional

   [ crl_ext ]
   authorityKeyIdentifier  = keyid:always
   ```

   Then, run these commands to create the CA certificate:

   ```bash
   mkdir -p myCA/private
   mkdir -p myCA/certs
   touch myCA/index.txt
   echo 1000 > myCA/serial
   echo 1000 > myCA/crlnumber

   openssl req -x509 -days 3650 -new -keyout myCA/private/ca_key.pem -out myCA/ca_cert.pem -config ca.conf
   openssl x509 -outform der -in myCA/ca_cert.pem -out myCA/ca_cert.der
   ```

2. **Client Certificate**  
   Create a client certificate using a similar process.

   `client-key.conf`:

   ```plaintext
   [ req ]
   default_bits = 4096
   default_md = sha256
   distinguished_name = DigilabClient
   req_extensions = req_ext
   x509_extensions = req_ext
   string_mask = utf8only
   prompt = no

   [ DigilabClient ]
   countryName = DE
   stateOrProvinceName = Bayern
   localityName = Munich
   organizationName = MTU
   commonName = client.local

   [ req_ext ]
   basicConstraints = CA:FALSE
   nsCertType = client, server
   keyUsage = nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign
   extendedKeyUsage = serverAuth, clientAuth
   nsComment = "Client 1 Certificate"
   subjectAltName = URI:urn:client.local:OPCUA:client.local,IP:127.0.0.1
   ```

   Generate the client certificate:

   ```bash
   openssl genpkey -algorithm RSA -out client_key.pem -pkeyopt rsa_keygen_bits:4096
   openssl req -new -key client_key.pem -out client.csr -config client-key.conf
   openssl ca -config ca.conf -in client.csr -out client_cert.pem -days 365 -extensions req_ext -extfile client-key.conf
   openssl x509 -outform der -in client_cert.pem -out client_cert.der
   ```

3. **Server Certificate**  
   Create the server certificate following the same approach.

   `server-key.conf`:

   ```plaintext
   [ req ]
   default_bits = 4096
   default_md = sha256
   distinguished_name = digilab_server
   req_extensions = req_ext
   x509_extensions = req_ext
   string_mask = utf8only
   prompt = no

   [ digilab_server ]
   countryName = DE
   stateOrProvinceName = Bayern
   localityName = Munich
   organizationName = MTU
   commonName = SimulationServer@Yoshi

   [ req_ext ]
   basicConstraints = CA:FALSE
   nsCertType = client, server
   keyUsage = nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign
   extendedKeyUsage= serverAuth, clientAuth
   nsComment = "Server Certificate"
   subjectAltName = URI:urn:Yoshi.lab.mtu-digilab.io:OPCUA:SimulationServer
   ```

   Generate the server certificate:

   ```bash
   openssl genpkey -algorithm RSA -out server_key.pem -pkeyopt rsa_keygen_bits:4096
   openssl req -new -key server_key.pem -out server.csr -config server-key.conf
   openssl ca -config ca.conf -in server.csr -out server_cert.pem -days 365 -extensions req_ext -extfile server-key.conf
   openssl x509 -outform der -in server_cert.pem -out server_cert.der
   ```

Best,
Gustavo

splatch, 2025-03-17:
I am not yet sure if I have a valid reproducer, but for sure I can't get client certificate to connect with Milo server using server certificate provisioned through your script. I keep getting `PlcIoException: Connection terminated by remote` and Milo is showing up:

```
java.security.cert.CertificateParsingException: java.io.IOException: subject key, java.security.InvalidKeyException: Invalid RSA public key
	at org.eclipse.milo.opcua.stack.core.util.CertificateUtil.decodeCertificates(CertificateUtil.java:123)
	at org.eclipse.milo.opcua.stack.core.util.CertificateUtil.decodeCertificate(CertificateUtil.java:83)
	at org.eclipse.milo.opcua.stack.core.util.CertificateUtil.decodeCertificate(CertificateUtil.java:72)
	at org.eclipse.milo.opcua.stack.core.channel.ServerSecureChannel.setRemoteCertificate(ServerSecureChannel.java:72)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerAsymmetricHandler.onOpenSecureChannel(UascServerAsymmetricHandler.java:194)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerAsymmetricHandler.decode(UascServerAsymmetricHandler.java:120)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:530)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:469)
	... 16 common frames omitted
Caused by: java.security.cert.CertificateParsingException: java.io.IOException: subject key, java.security.InvalidKeyException: Invalid RSA public key
```

Do you get any client/server stack traces to share?

Nguyen-BangVu, 2025-06-03:
I have some other issue
when I use a 2048 bit cert for the server side (prosys simiulation) and a 4096 bit cert from the client side (plc4x). 
The connection can not even be established.

java.util.concurrent.CompletionException: java.lang.RuntimeException: javax.crypto.IllegalBlockSizeException: Data must not be longer than 214 bytes
	at java.base/java.util.concurrent.CompletableFuture.encodeThrowable(CompletableFuture.java:314)
	at java.base/java.util.concurrent.CompletableFuture.completeThrowable(CompletableFuture.java:319)
	at java.base/java.util.concurrent.CompletableFuture$UniCompose.tryFire(CompletableFuture.java:1081)
	at java.base/java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:506)
	at java.base/java.util.concurrent.CompletableFuture.complete(CompletableFuture.java:2079)
	at org.apache.plc4x.java.opcua.context.Conversation.lambda$4(Conversation.java:198)
	at java.base/java.util.function.Consumer.lambda$andThen$0(Consumer.java:65)
	at org.apache.plc4x.java.spi.Plc4xNettyWrapper.decode(Plc4xNettyWrapper.java:183)
	at io.netty.handler.codec.MessageToMessageCodec$2.decode(MessageToMessageCodec.java:81)
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:88)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:103)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:346)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:318)
	at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:440)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:788)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:724)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:650)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:997)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.base/java.lang.Thread.run(Thread.java:829)
Caused by: java.lang.RuntimeException: javax.crypto.IllegalBlockSizeException: Data must not be longer than 214 bytes
	at org.apache.plc4x.java.opcua.context.BaseEncryptionHandler.encodeMessage(BaseEncryptionHandler.java:121)
	at org.apache.plc4x.java.opcua.context.EncryptionHandler.encodeMessage(EncryptionHandler.java:73)
	at org.apache.plc4x.java.opcua.context.Conversation.request(Conversation.java:237)
	at org.apache.plc4x.java.opcua.context.Conversation.requestChannelOpen(Conversation.java:204)
	at org.apache.plc4x.java.opcua.context.SecureChannel.onConnectOpenSecureChannel(SecureChannel.java:209)
	at org.apache.plc4x.java.opcua.context.SecureChannel.lambda$1(SecureChannel.java:153)
	at java.base/java.util.concurrent.CompletableFuture$UniCompose.tryFire(CompletableFuture.java:1072)
	... 35 more

In theory everything should work with 2 different sized certs doesn't it

Nguyen-BangVu, 2025-06-07:
Regarding the 4096 client, 4096 server problem. I put a thread into the prosys forum. 
I quote:  "At the moment, Simulation Server supports only using 2048 bit certificates as its Application Instance Certificate."

chrisdutz, 2026-08-15:
Well ... please cut that ... it wasn't easy, but as I think it's important I just added support for the trust chain. Tested with my own ssl.com certificate. 

Closing this as addressed. The thread accumulated several different problems, so a summary of where each stands:

The mixed key-length failure (@Nguyen-BangVu — IllegalBlockSizeException: Data must not be longer than 214 bytes with a 4096-bit client against a 2048-bit server) is fixed: the asymmetric block sizes are now derived from the remote certificate instead of the local key. 214 bytes is exactly the OAEP plaintext limit for a 2048-bit key, which is what the client should have been sizing against.

The 4096/4096 case against Prosys turned out to be a server limitation, as @Nguyen-BangVu established: Simulation Server only supports 2048-bit application instance certificates.

The key size itself is not the problem. The test suite now connects with a 4096-bit client certificate across every secured security policy against Eclipse Milo, and a 4096-bit key store loads cleanly.

Several real defects turned up while investigating, any of which could produce the original "an error occurred verifying security":
- the certificate the driver generates when no key-store-file is configured was signed with SHA-1, marked as a CA (basicConstraints=true plus keyCertSign), carried a 40-octet and often negative serial number, and had no subject key identifier — each of those alone is enough for a server with strict checks to refuse it
- a CA-signed client certificate was sent without the certificates that signed it, so a server trusting only the issuing CA had nothing to build a path from. OPC UA Part 6 allows appending them to SenderCertificate, and we now do
- the key store entry used was whatever alias came first, which for the CA-plus-client key stores this setup produces could hand the CA's entry to the server. A missing private key or a non-RSA key now fails immediately with a message naming the entry, rather than as an opaque handshake failure
- generated-key-size was added for anyone needing the generated certificate above 2048 bits

@gustavoschewinski  — if this still occurs on current develop, please reopen with the output of:

export PLC4X_OPCUA_TEST_KEYSTORE_PASSWORD='...'
mvn test -Dtest=RealKeyStoreManualTest -pl plc4j/drivers/opcua \
  -Dplc4x.opcua.test.keystore=/absolute/path/to/your.p12

It prints the key algorithm, key size, signature algorithm and full chain of your client identity — which is what we'd need to tell whether the certificate or the driver is at fault.                                                                                                     

@splatch — your Invalid RSA public key reproduction attempt is the one thread none of the above explains, since it's the server failing to parse the client certificate's public key. If that still reproduces with a tutorial-provisioned certificate, it deserves its own issue.

closed 2026-08-15 by chrisdutz

splatch, 2026-08-28:
@chrisdutz I would blame bug with swapped client/server certificate in driver for most of X509 related issues. With recent fixes most of cert issues should be gone.