apache/plc4x #2019: [Bug]: IoTDB documentation refers to missing example

slawr, 2025-02-21:
### What happened?

The Apache IoTDB user documentation https://plc4x.apache.org/plc4x/latest/users/integrations/apache-iotdb.html links to an example https://github.com/apache/plc4x/tree/develop/plc4j/examples/hello-integration-iotdb which does not exist.

I can not find the file or something close to it in the sources so am unable to provide a fix.

### Version

v0.12.0

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

klaus321901, 2025-10-02:
I’d like to contribute to this issue and work on fixing the documentation link.

chrisdutz, 2026-08-16:
Fixed - and you were right that it could not be found.

The examples moved out of the main repository into `plc4x-extras`, which is why the path did not
exist here. The IoTDB page now links to
https://github.com/apache/plc4x-extras/tree/develop/plc4j/examples/hello-integration-iotdb
which is where that example actually lives.

Closing as fixed.

closed 2026-08-16 by chrisdutz