apache/plc4x #2023: [Bug]: Inaccurate PLC Protocol Compatibility Matrix Webpage

dtober-softeon, 2025-02-26:
### What happened?

# Description
The latest [language and protocol support page](https://plc4x.apache.org/plc4x/latest/users/protocols/index.html) does not accurately reflect current compatibilities. Issue originally raised as a [discussion thread](https://github.com/apache/plc4x/discussions/2001) on the same topic. 

Out-dated compatibility information is an active deterrent for potential users and contributors.

# Requested changes
Please update the respective tables on the page to accurately represent Features, Languages, and Protocols that are supported by PLC4X.

### Version

v0.12.0

### Programming Languages

- [x] plc4j
- [x] plc4go
- [x] plc4c
- [x] plc4net

### Protocols

- [x] AB-Ethernet
- [x] ADS /AMS
- [x] BACnet/IP
- [x] CANopen
- [x] DeltaV
- [x] DF1
- [x] EtherNet/IP
- [x] Firmata
- [x] KNXnet/IP
- [x] Modbus
- [x] OPC-UA
- [x] S7

klaus321901, 2025-10-02:
Hi, I’m new to Apache PLC4X and would like to work on this issue.

chrisdutz, 2026-08-16:
Fixed.

closed 2026-08-16 by chrisdutz