apache/plc4x #2026: [Bug]: It seems to be impossible to run the plc4py build with skip tests

chrisdutz, 2025-02-27:
### What happened?

When building PLC4X it sometimes is usefull to build all parts locally, skipping all tests. IntelliJ has a checkbox in the maven panel, which I think passes the "-DskipTests" to the maven build. If this is set, the plc4py build fails.

In our build README we tell people to build with skipTests to skip the tests, if the build fails we might have problems with the next release.

### Version

0.13.0-SNAPSHOT

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2025-02-27:
It seems that when "skipTests" is enabled, there are two modules missing: 
- black 
- deptry

If I comment out the two executions requiring them (python-black, python-dependency-check), then I am able to build the project with "skipTests".  It seems that when not set, these dependencies are pulled in in the python-test-compile step ... at least can I see them being mentioned in the log then.

Just thought this might help in tracking down the issue.

sruehl, 2025-03-28:
I wonder if https://github.com/apache/plc4x/commit/05d74143784b4af014286fb4dc128904c37ade0e fixes it and this issue was never closed?

Abdylreshit, 2026-04-24:
Confirmed — this is already fixed by commit [05d7414](https://github.com/apache/plc4x/commit/05d74143784b4af014286fb4dc128904c37ade0e) (Feb 2025), which removed `<skip>${skipTests}</skip>` from the `python-test-compile` execution in `plc4py/pom.xml`. That execution installs the `.[dev]` extras (bringing in `black` and `deptry`), so they are now available for the `python-black` and `python-dependency-check` steps regardless of whether `-DskipTests` is set.

Verified locally on current `develop` (commit 6f748aa958):

```
mvn install -DskipTests -P with-python -pl plc4py -am
...
[INFO] PLC4X .............................................. SUCCESS [01:36 min]
[INFO] PLC4Py ............................................. SUCCESS [ 49.403 s]
[INFO] BUILD SUCCESS
```

This issue can be closed.

m13v, 2026-04-25:
this is the classic polyglot maven-python build coupling problem. ran into the same shape last year on a mixed java/python IIoT project where -DskipTests propagated to the python module and broke the setuptools step because the test scaffolding was also generating fixture data the wheel build depended on. the fix was splitting -DskipTests (java) from a separate -Dpython.skip.tests flag in the plc4py pom so the python build could keep running its build-time codegen while still skipping pytest. otherwise every IDE user who clicks the maven skipTests checkbox keeps getting bitten.

sruehl, 2026-08-18:
Verified on current develop: this was fixed by 05d74143784b4af014286fb4dc128904c37ade0e, which removed <skip>${skipTests}</skip> from the python-test-compile execution in plc4py/pom.xml. That execution runs pip install .[dev], which provides black and deptry, so the python-black and python-dependency-check executions now work regardless of -DskipTests. Only the actual pytest run (python-test) is skipped, which is the intended behavior.

Reproduced locally:
```
./mvnw install -DskipTests -P with-python -pl plc4py -am
...
[INFO] PLC4X .............................................. SUCCESS
[INFO] PLC4Py ............................................. SUCCESS
[INFO] BUILD SUCCESS
```

Regarding the suggestion of a separate -Dpython.skip.tests flag: that's not needed anymore, since the dev-dependency install is already decoupled from skipTests — and it would stop the IDE's skipTests checkbox from skipping pytest.

Closing as fixed.

closed 2026-08-18 by sruehl