apache/plc4x #2027: [Bug]: s7-1500, read, xxx  invalid

anganing, 2025-03-01:
### What happened?

hello,

I use plc4x v0.10.0 java, to read a plc address, but erro.

all error!

[image]

[image]

[image]

### Version

v0.10.0

### Programming Languages

- [ ] plc4c
- [ ] plc4go
- [x] plc4j
- [ ] plc4net
- [ ] plc4py

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] C-Bus
- [ ] CANopen
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] IEC-69870
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] Profinet
- [x] S7

chrisdutz, 2025-03-01:
Please try the latest released version, which is 0.12.0 or our current unreleased version 0.13.0-SNAPSHOT first.

chrisdutz, 2025-03-01:
We're a tiny team maintaining plc4x in our free time, we can't provide support for all historic versions, so we're concentrating on the last released version and it might even be that it's already fixed, but not yet released. You can help us by filtering out possible false positives.

chrisdutz, 2026-08-15:
Closing this as stale. It was reported against 0.10.0, and both the SPI and the S7 driver have been rewritten since (the s7 and s7-light drivers are now one driver on a Netty-free SPI), so the code this describes no longer exists. There's also nothing in the report we can act on — the screenshots don't tell us the tag addresses, the error codes or the connection string, and there's been no response to the request to retest on a newer version.                                                                                                                                                     

If you still see this, please open a fresh issue against current develop with the tag addresses as text and the log output. Two things have improved since that will make it much easier to diagnose: an unreadable address now comes back as INVALID_ADDRESS for that tag alone while the rest of the request still returns data, and the driver now logs which address was rejected and why.

For an S7-1500 it's also worth checking the two usual suspects first: the data block must have optimized block access disabled, and PUT/GET communication must be permitted in the CPU's protection settings.

closed 2026-08-15 by chrisdutz