apache/plc4x #2045: [Bug]: modbus read timeout, CachedPlcConnectionManager will not create a new connect

jackxuhaijun, 2025-03-27:
### What happened?

```
    @SneakyThrows
    public LiftReadData readData(Area area) {
        LiftReadData readData = null;
        try {
            readData = entityManager.read(LiftReadData.class, area.getUrl());
        } catch (Exception e) {
            log.info("PLC4X {} read exception", area);
            throw new RuntimeException(e);
        }
        log.info("PLC4X {} read {}", area, readData);
        return readData;
    }
```
I am using the Modbus-TCP protocol with a read timeout of 1 second. The on-site network frequently experiences packet loss, causing the connection to remain in the TIME_WAIT state. When attempting to reconnect, no new connection is created, and there is no network communication at all. Instead, a timeout exception is thrown immediately.

### Version

v.0.12.0

### Programming Languages

- [ ] plc4c
- [ ] plc4go
- [x] plc4j
- [ ] plc4net
- [ ] plc4py

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] C-Bus
- [ ] CANopen
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] IEC-69870
- [ ] KNXnet/IP
- [x] Modbus
- [ ] OPC-UA
- [ ] Profinet
- [ ] S7

jackxuhaijun, 2025-03-27:
this is the exception
```
java.lang.RuntimeException: org.apache.plc4x.java.opm.OPMException: Unexpected error during processing
	at com.zikoo.rcs.server.lift.LiftService.getWriteData(LiftService.java:113) ~[zk-lift-1.1.0.jar!/:?]
	at com.zikoo.rcs.server.lift.LiftService$$FastClassBySpringCGLIB$$8ad9a247.invoke(<generated>) ~[zk-lift-1.1.0.jar!/:?]
	at org.springframework.cglib.proxy.MethodProxy.invoke(MethodProxy.java:218) ~[spring-core-5.3.27.jar!/:5.3.27]
	at org.springframework.aop.framework.CglibAopProxy.invokeMethod(CglibAopProxy.java:386) ~[spring-aop-5.3.27.jar!/:5.3.27]
	at org.springframework.aop.framework.CglibAopProxy.access$000(CglibAopProxy.java:85) ~[spring-aop-5.3.27.jar!/:5.3.27]
	at org.springframework.aop.framework.CglibAopProxy$DynamicAdvisedInterceptor.intercept(CglibAopProxy.java:704) ~[spring-aop-5.3.27.jar!/:5.3.27]
	at com.zikoo.rcs.server.lift.LiftService$$EnhancerBySpringCGLIB$$7fa53755.getWriteData(<generated>) ~[zk-lift-1.1.0.jar!/:?]
	at com.zikoo.rcs.server.lift.LiftServer.getWriteData(LiftServer.java:82) ~[zk-lift-1.1.0.jar!/:?]
	at com.zikoo.rcs.server.lift.LiftServer.getWriteData(LiftServer.java:25) ~[zk-lift-1.1.0.jar!/:?]
	at com.zikoo.rcs.server.lift.LiftServerHandler.handle(LiftServerHandler.java:22) ~[lift-api-1.1.0.jar!/:?]
	at com.zikoo.rcs.api.ServerHandler.internalInvoke(ServerHandler.java:219) ~[core-2.0.0.jar!/:?]
	at com.zikoo.rcs.api.ServerHandler.run(ServerHandler.java:150) ~[core-2.0.0.jar!/:?]
	at org.springframework.scheduling.support.DelegatingErrorHandlingRunnable.run(DelegatingErrorHandlingRunnable.java:54) ~[spring-context-5.3.27.jar!/:5.3.27]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) ~[?:?]
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) ~[?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
	at java.lang.Thread.run(Thread.java:834) ~[?:?]
Caused by: org.apache.plc4x.java.opm.OPMException: Unexpected error during processing
	at org.apache.plc4x.java.opm.PlcEntityInterceptor.refetchAllFields(PlcEntityInterceptor.java:255) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityManager.connect(PlcEntityManager.java:171) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityManager.connect(PlcEntityManager.java:126) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityManager.read(PlcEntityManager.java:106) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at com.zikoo.rcs.server.lift.LiftService.getWriteData(LiftService.java:109) ~[zk-lift-1.1.0.jar!/:?]
	... 18 more
Caused by: org.apache.plc4x.java.opm.OPMException: Timeout during fetching values
	at org.apache.plc4x.java.opm.PlcEntityInterceptor.getFromFuture(PlcEntityInterceptor.java:563) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityInterceptor.getPlcReadResponse(PlcEntityInterceptor.java:539) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityInterceptor.refetchAllFields(PlcEntityInterceptor.java:237) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityManager.connect(PlcEntityManager.java:171) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityManager.connect(PlcEntityManager.java:126) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityManager.read(PlcEntityManager.java:106) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at com.zikoo.rcs.server.lift.LiftService.getWriteData(LiftService.java:109) ~[zk-lift-1.1.0.jar!/:?]
	... 18 more
Caused by: java.util.concurrent.TimeoutException
	at java.util.concurrent.CompletableFuture.timedGet(CompletableFuture.java:1886) ~[?:?]
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2021) ~[?:?]
	at org.apache.plc4x.java.opm.PlcEntityInterceptor.getFromFuture(PlcEntityInterceptor.java:556) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityInterceptor.getPlcReadResponse(PlcEntityInterceptor.java:539) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityInterceptor.refetchAllFields(PlcEntityInterceptor.java:237) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityManager.connect(PlcEntityManager.java:171) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityManager.connect(PlcEntityManager.java:126) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at org.apache.plc4x.java.opm.PlcEntityManager.read(PlcEntityManager.java:106) ~[plc4j-opm-0.11.0.jar!/:0.11.0]
	at com.zikoo.rcs.server.lift.LiftService.getWriteData(LiftService.java:109) ~[zk-lift-1.1.0.jar!/:?]
	... 18 more
```

jackxuhaijun, 2025-03-28:
> Indeed this seems to be a common problem with various of our drivers ... gracefully handling disconnects, network problems etc. and allowing a quick recovery.
Thank you very much for your prompt reply！
Based on my observations, when using DefaultPlcConnectionManager, a new connection is established after each read operation. In contrast, CachedPlcConnectionManager reuses connections, but if the socket is in the TIME_WAIT state, no communication occurs. In this case, a new CachedPlcConnectionManager needs to be created. However, the old CachedPlcConnectionManager is not properly released, leading to thread leaks and port occupation. If this happens frequently, it can result in too many threads, eventually causing an OOM (Out of Memory) issue.
So now I don’t know the best way to use it. Do you have any suggestions?

chrisdutz, 2025-03-28:
Well ... I would say that creation of new connections is not related to the read operation, but to the call of "getConnection" ... the DefaultDriverManager simply creates a new connection and the CachedDriverManager uses the DefaultDriverManager to create a connection, but then keeps it active.

But yes ... I guess all of our drivers could use a dedicated overwork to detect the disconnected state as well as terminate waiting futures as soon as a disconnect happens. I think currently if you start a read operation and then pull the plug, that the readRequest's future will not be cancelled, even if the connection allmost instantly is notified. In this case the Future will timeout.

However ... a lot of the complexity is related to Netty stuff ... and implementing something that instantly terminates would also require quite a rewrite. 

Also are we running super low on people still wanting to invest their time here without ... we are currently trying to find ways to incentivise active addressing of bugs and refactorings (Still waiting for final approval of Apache Trademarks team before we can make this official) ... but if your company benefits of Apache PLC4X you already might want to spend a few minutes on if you would be willing to financially support the project.