apache/plc4x #2049: [Feature Request]: omron  finstcp/insudp      protocol support

boyucjz, 2025-04-02:
### What would you like to happen?

omron  finstcp/insudp      protocol support

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2025-04-02:
Implementing a new protocol/driver usually requires quite a bit of invest for time and importantly equipment. Are you or anyone you know of willing to invest into such a driver? (Providing Hard- and/or Software, financial incentives for investing the time to implement such a driver)

boyucjz, 2025-04-02:
Yeah, I want to make that happen

chrisdutz, 2025-04-02:
Hi,

we just recently created an opencollective collective "Friends of Toddy" to handle the financing of things as an Apache Project is not allowed to manage such things because of it's non-profit charity status.

Admittedly I have no idea on how complicated the protocol is and to how much funds would be required to make it happen. Also maybe others want to do it for free ;-) 

https://opencollective.com/friends-of-toddy

If you have some information on the project, I could possibly add a "Project" on the collective page and attach a price-tag to it.

boyucjz, 2025-04-02:
I am a programmer and would very much like to find interested people here to implement a piece

chrisdutz, 2025-04-04:
Sorry for the late reply ... had to finish some other things ... I think. i really should document the steps I do to create:
a) A new driver for an existing language
b) A new language module

Please be a bit patient ... but as there are no funds coming in, I'll not work on any issues, features etc., so I've to time to write more ;-)

chrisdutz, 2025-04-06:
Ok ... here comes a first "birds-eye-view" of how I start a new driver. I guess getting that to work will probably keep you occupied until I manage to continue it a bit further.

chrisdutz, 2025-04-11:
Ups ... forgot to post the link: https://plc4x.apache.org/plc4x/pre-release/developers/tutorials/writing-driver.html