apache/plc4x #2050: [Bug]: Response failure with S7315-2PN/DP. The last packet is not processed.

glcj, 2025-04-06:
### What happened?

When requesting a very long structure, the optimizer correctly partitions the requests but ignores the last response packet.

  Request:          readrequest.addTagAddress("TEST", "%DB45:0:USINT[896]"); //(02) 

[s7315.zip](https://github.com/user-attachments/files/19623853/s7315.zip)

### Version

v0.13.0-SNAPSHOT

### Programming Languages

- [ ] plc4c
- [ ] plc4go
- [x] plc4j
- [ ] plc4net
- [ ] plc4py

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] C-Bus
- [ ] CANopen
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] IEC-69870
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] Profinet
- [x] S7

glcj, 2025-04-07:
List of bytes returned by the optimizer encapsulated in an optimizer object.

[image]

Without optimization it returns a PlcList

[image]

chrisdutz, 2025-04-07:
It's this using the default optimizer or the new block based one? And does it affect the s7 or s7-light driver?

imagen.png (view on web)<https://github.com/user-attachments/assets/4a310e13-8193-44d9-9601-9f8ef052d4bd>

imagen.png (view on web)<https://github.com/user-attachments/assets/8ee108f0-ad27-4630-abbb-05d8d07b9b2a>

imagen.png (view on web)<https://github.com/user-attachments/assets/4a310e13-8193-44d9-9601-9f8ef052d4bd>

imagen.png (view on web)<https://github.com/user-attachments/assets/8ee108f0-ad27-4630-abbb-05d8d07b9b2a>

glcj, 2025-04-07:
Hi,

The default optimizer does its job, but the request returns an inner class. I have two response types for the same request.

chrisdutz, 2025-04-07:
As I don't seem to quite understand your response: So the default s7 optimizer (not the new version I built) is generally doing, what it should, but it's returning an inner class (The modifyable list) and this is missing the content of the last packet?

glcj, 2025-04-07:
Hello,
After continuing the processing, the Optimizer performs its work correctly.
The problem is in the returned object.

For a request that is smaller than the maximum PDU, my program expects a "PlcList," which works without problems.

When the Optimizer comes in, that is, when the returned data is larger than the size of a PDU, the returned type does not correspond to a "PlcList," but rather "S7Optimizer.PlcListModifiable."

I think the returned "PlcValue" should be the expected type ("PlcList") and not a type not included in the API/SPI.

This difference mainly affects me with isolated objects (OSGi).

I'm modifying the application to use "PlcValue."

chrisdutz, 2025-04-08:
But the value it returns is a PlcValue. Yes it's not a PlcList, but PlcList itself is not an API type. A PlcList implements the same interface as any PlcValue. And as the modifyable list in the optimizer extends the PlcValueAdapter this should too.

chrisdutz, 2025-04-08:
So I think the fully API compliant way to do it, would be to check if it's PlcValue type and if this is a list and then to call the getList() method.

glcj, 2025-04-09:
Hello,

I'm going to close this bug as it's more a matter of interpretation.

Ultimately, my problem is consistently solved by reading the data directly from the byte array with "%DB45:0:BYTE[896]" or "%DB45:0:RAW_BYTE_ARRAY[896]", returning a "PlcRawByteArray".

If variable-length data types are requested, the PlcValue interpretation is not straightforward, so it must be evaluated at the application level.

Thanks,

closed 2025-04-09 by glcj