apache/plc4x #2060: [Bug]: Modbus: Reading Coils as an array returns only the first value

IsmoLeszczynski, 2025-04-25:
### What happened?

When defining a `ModbusTag` with array size > 1, the `ModbusOptimizer.processReadResponses(...)` `ModbusTagCoil` logic returns only the first bit as the result. The array logic works correctly for registers, but the customized coil logic is missing the array handling:

https://github.com/apache/plc4x/blob/a7ab2cf08c8b06b44c0d1b7ff8b71813d6a4966b/plc4j/drivers/modbus/src/main/java/org/apache/plc4x/java/modbus/base/optimizer/ModbusOptimizer.java#L210-L216

What it should do is check for the for the `modbusTag.getNumberOfElements()` and generate the `PlcList` when needed. My current development version fixes this as follows:

```java
if (modbusTag.getNumberOfElements() > 1) {
  PlcList bitValues = new PlcList();
  for (int i = 0; i < modbusTag.getNumberOfElements(); i++) {
    int bitPosition = modbusTag.getAddress() - response.startingAddress + i;
    boolean isBitSet = getIsBitSet(bitPosition, responseData);
    bitValues.add(new PlcBOOL(isBitSet));
  }
  values.put(tagName, new DefaultPlcResponseItem<>(PlcResponseCode.OK, bitValues));
} else {
  int bitPosition = modbusTag.getAddress() - response.startingAddress;
  boolean isBitSet = getIsBitSet(bitPosition, responseData);
  values.put(tagName, new DefaultPlcResponseItem<>(PlcResponseCode.OK, new PlcBOOL(isBitSet)));
}
```
```java
private static boolean getIsBitSet(int bitPosition, byte[] responseData) {
  int bytePosition = bitPosition / 8;
  int bitPositionInByte = bitPosition % 8;
  boolean isBitSet = (responseData[bytePosition] & (1 << bitPositionInByte)) != 0;
  return isBitSet;
}
```

### Version

v0.13.0-SNAPSHOT

### Programming Languages

- [ ] plc4c
- [ ] plc4go
- [x] plc4j
- [ ] plc4net
- [ ] plc4py

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] C-Bus
- [ ] CANopen
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] IEC-69870
- [ ] KNXnet/IP
- [x] Modbus
- [ ] OPC-UA
- [ ] Profinet
- [ ] S7
- [ ] S7-light

IsmoLeszczynski, 2025-09-02:
I made a PR for this, #2192 , any chance to have that reviewed? Admittedly, I bundled two issues in the same PR, if that's a problem I can split the requests. Thanks in advance!

chrisdutz, 2025-09-02:
Öhm ... could you please double-check it' not fixed in the current version?