apache/plc4x #2061: [Feature Request]: Modbus: WORD, DWORD and LWORD array support

IsmoLeszczynski, 2025-04-25:
### What would you like to happen?

While having a look at the MSpec for Modbus, I noted that for some reason the WORD-variants of data types don't have List variants, unlike every other data type.

https://github.com/apache/plc4x/blob/a7ab2cf08c8b06b44c0d1b7ff8b71813d6a4966b/protocols/modbus/src/main/resources/protocols/modbus/modbus.mspec#L339-L347

Is there any reason for this, or could they just be added to the MSpec? I ran in to this when implementing reading of single bits from multiple registers - my implementation used WORD with array dimensions as the data type for the ModbusTag when reading the registers, and I noticed that I only ever got a single register as the result. The immediate fix was to change to INT, but I don't see why the WORD variants wouldn't have array support as well.

### Programming Languages

- [x] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [x] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2026-08-18:
I guess the main reason for leaving them out is that usually the case name decides which type of PlcValue is being used. With these types there comes an issue: we read them as for example an uint 16 ... that would result in an array of PlcUINT and not an array of PlcWORD.

I think it might make sense to generally refactor the mspec bing used in modbus to not support the array types, but to move that into the driver (like with other drivers). Then this should be greatly simplified.

chrisdutz, 2026-08-18:
Your diagnosis was exactly right - `WORD`, `DWORD` and `LWORD` had no list variant in the
`DataItem` type switch, so a request for several of them matched the single-value case and only
the first register was parsed. The request itself was fine: the driver did ask the device for the
full range, the decoding just stopped after one value. Hence the fallback to `INT` working.

To answer the actual question - yes, there was a technical reason, and it is why this took a
rewrite rather than three added lines. A list case in the mspec generates an array over the
underlying field, so a `WORD` list would have been `[array uint 16 value ...]` and would have
returned a list of `PlcUINT`. There was no way to express "a list whose elements are `PlcWORD`".
Adding the missing cases would only have swapped a truncated read for a wrongly typed one.

So the array handling was moved out of the mspec altogether. `DataItem` now describes exactly one
value of a type, and the register layout Modbus adds on top - a value narrower than a register
being padded to fill one, several of them packed two per register, a trailing pad when an odd
count leaves the last register half full - lives in a `ModbusRegisterCodec` in the Java, Go and
Python drivers. An array is simply N calls to the single-value parser, which means every element
goes through the `WORD` branch and you get `PlcWORD` values back, not `PlcUINT`. It also means no
data type can end up without array support again.

Thanks for the clear report - it pointed straight at the inconsistency, and at a design problem
behind it.

Closing as implemented.

closed 2026-08-18 by chrisdutz

IsmoLeszczynski, 2026-08-19:
Awesome, thanks for the update! I'm working right now on checking out the updated drivers, I'll try this out while doing a wider pass this week.