apache/plc4x #2088: [Feature Request]: ads connection problem

bluecybermzq, 2025-05-15:
### What would you like to happen?

I am trying to use the plc4j to conncect a Beckhoff plc unsing ads, but cannot connect to the twincat3 simulator. my code is like this:
    @Test
    public void plc4xAdsTest() throws PlcConnectionException {

        String targetAmsNetId =  "192.168.159.1.1.1";
        Integer targetAmsPort = 851;
        String sourceAmsNetId = "192.168.101.45.1.1";
        Integer sourceAmsPort = 65534;
        String timeout = "5000";
        String urlFormat = "ads:tcp://192.168.159.1:851?target-ams-net-id={}&target-ams-port={}&source-ams-net-id={}&source-ams-port={}&timeout-request={}&tcp.default-timeout=6000";
        String connectionString = StrUtil.format(urlFormat,targetAmsNetId,targetAmsPort,sourceAmsNetId,sourceAmsPort,timeout);

        try {

            PlcConnection plcConnection = PlcDriverManager
                    .getDefault()
                    .getConnectionManager()
                    .getConnection(connectionString);
            //2， readSupported?
            if (!plcConnection.getMetadata().isReadSupported()){
                System.out.println("This connection doesn't support reading.");
                return;
            }
            //3, writeSupported?
            if (!plcConnection.getMetadata().isWriteSupported()){
                System.out.println("This connection doesn't support writing.");
                return;
            }
            if (plcConnection.isConnected()){
                System.out.println(">>>>>>>>>ads connected。");
            }
            plcConnection.close();

        } catch (Exception e) {
            throw new PlcConnectionException(e);
        }
    }

and get the connection refused exception .

[image]

but I use Beckhoff's dll to connect the simulator, it is fine:
    @Test
    public void plcAdstest2(){
        AmsAddr addr = new AmsAddr();
        JNIByteBuffer hdlBuff = new JNIByteBuffer(4);
        JNIByteBuffer symBuff = new JNIByteBuffer(Convert.StringToByteArr("MAIN.int1"));
        JNIByteBuffer dataBuff = new JNIByteBuffer(2);

        long nPort = AdsCallDllFunction.adsPortOpen();
        long nErr = AdsCallDllFunction.getLocalAddress(addr);
        addr.setNetIdString("192.168.159.1.1.1");
        addr.setPort(851);
        if (nErr!=0){
            System.out.println("ERROR: open Communication: 0x");
        }

        nErr = AdsCallDllFunction.adsSyncReadReq(addr,0x4040,0x5DFE8,2,dataBuff);
        if (nErr!=0){                
            System.out.println("ERROR: Get handle: 0x");
            int data = Convert.ByteArrToInt(dataBuff.getByteArray());
            int hdl = Convert.ByteArrToInt(hdlBuff.getByteArray());
            int sym = Convert.ByteArrToInt(symBuff.getByteArray());
            System.out.println("sym:"+sym+ "hdl:"+hdl+"data:" +data);
            System.out.println("nErr:" + nErr);
        }else {
            int intVal = Convert.ByteArrToInt(dataBuff.getByteArray());
            System.out.println("PLCVar value: "+ intVal);
        }

    }

it is my first trying to use plc4x and twincat3 , is there any tricks in connection string or any tricks in twincat3? 

### Programming Languages

- [x] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [x] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

bluecybermzq, 2025-05-15:
my plc4x version is 0.12.0, twincat3 version is 3.1.4024.29. java version is JDK17.
thanks for any help!

UserwithnoUsername, 2025-07-18:
String urlFormat = "ads:tcp://192.168.159.1:851?

851 is the ADS Internal Port. I Think you need 48898 here.

https://infosys.beckhoff.com/index.php?content=../content/1031/ipc_security_win7/11019143435.html&id=

chrisdutz, 2026-08-16:
Can we close this issue? The driver has proven to be able to connect, so it's not a connection issue or anything we can fix here. I assume you need to configure the AMS Router correctly.

bluecybermzq, 2026-08-16:
yes, it may be the configuration problem of AMS Router. now we should close the issue.

closed 2026-08-16 by chrisdutz